BerkeleyAutomation/gqcnn #5: GQ Image Wise Config JSON Not Found

ashwinreddy, 2017-07-04:
Hello,
I have installed `autolab_core`, `perception`, and `gqcnn` using a Conda environment with Python 2.7.13 on macOS. However, when I try to run the example policy:

```
$ python2.7 examples/policy.py --config_filename ./cfg/examples/policy.yaml                                                    areddy@panzer
WARNING:root:Failed to import geometry msgs in rigid_transformations.py.
WARNING:root:Failed to import ros dependencies in rigid_transforms.py
WARNING:root:autolab_core not installed as catkin package, RigidTransform ros methods will be unavailable
WARNING: AUTOLab Perception Module not installed as Catkin Package. ROS msg conversions will not be available for Perception wrappers
WARNING: AUTOLab Perception module not installed as Catkin Package. ROS msg conversions will not be available for Perception wrappers.
WARNING:root:Unable to import pylibfreenect2. Python-only Kinect driver may not work properly.
WARNING:root:Unable to import openni2 driver. Python-only Primesense driver may not work properly
WARNING:root:Failed to import ROS in primesense_sensor.py. ROS functionality not available
WARNING:root:primesense_sensor.py not installed as catkin package. ROS functionality not available.
DEBUG:PIL.PngImagePlugin:STREAM 'IHDR' 16 13
DEBUG:PIL.PngImagePlugin:STREAM 'IDAT' 41 65536
Traceback (most recent call last):
  File "examples/policy.py", line 143, in <module>
    policy = CrossEntropyAntipodalGraspingPolicy(policy_config)
  File "/Users/areddy/Research/DexNet/gqcnn/gqcnn/policy.py", line 377, in __init__
    GraspingPolicy.__init__(self, config)
  File "/Users/areddy/Research/DexNet/gqcnn/gqcnn/policy.py", line 104, in __init__
    self._gqcnn = GQCNN.load(self._gqcnn_model_dir)
  File "/Users/areddy/Research/DexNet/gqcnn/gqcnn/neural_networks.py", line 71, in load
    with open(config_file) as data_file:    
IOError: [Errno 2] No such file or directory: '/home/user/data/models/GQ-Image-Wise/config.json'
```

visatish, 2017-07-04:
This error usually means that the path to the model, in your case '/home/user/data/models/GQ-Image-Wise/', is incorrect t. Can you check if that directory exists in your system and contains a config.json file.

ashwinreddy, 2017-07-05:
The directory does not exist. Do I need to download this folder from somewhere? I thought that installing the dependencies would be enough to run the policy example.

visatish, 2017-07-05:
Running the policy requires a GQ-CNN model. These models can be downloaded from here: https://berkeley.app.box.com/s/szbchyt3tou9e4ct6dz8c5v99vhx0s84. Once you have the model downloaded, change the policy/gqcnn_model parameter in the policy.yaml config file to the location of the model on your machine.

closed 2017-07-16 by jeffmahler