BerkeleyAutomation/gqcnn #6: XXX not installed as catkin package

shinshiner, 2017-07-05:
Hello, I have followed the instructions [here](https://berkeleyautomation.github.io/gqcnn/install/install.html#ros-installation) to install gqcnn as a ros package. However, when I execute 

`python tools/training.py`

it posts these warnings:

```
WARNING:root:autolab_core not installed as catkin package, RigidTransform ros methods will be unavailable

WARNING: AUTOLab Perception Module not installed as Catkin Package. ROS msg conversions will not be available for Perception wrappers

WARNING:root:Unable to import pylibfreenect2. Python-only Kinect driver may not work properly.

WARNING:root:Unable to import openni2 driver. Python-only Primesense driver may not work properly

WARNING:root:primesense_sensor.py not installed as catkin package. ROS functionality not available.

```

I have followed the instructions [here](https://berkeleyautomation.github.io/autolab_core/install/install.html#ros-installation) to install autolab_core as a ros package.

Besides, I find the [perception module](https://berkeleyautomation.github.io/perception/install/install.html) does not have ros installation. How to install it as a catkin package to fix the second warning?

By the way, could anyone explains the last three warnings? It seems that I should install some more packages?

Thank you!

visatish, 2017-07-05:
1) The installation documentation for the perception module is not up-to-date, it can be installed as a ROS package by following the same steps as for the other AUTOLab modules.(catkin make + source setup.bash)
2) The warnings mean that the ros install was not successful. Did you make sure to source the setup.bash for your catkin workspace after running catkin make? If that doesn't work could you post the output of your catkin make for reference?
3) openni2 and pylibfreenect2 are extra packages that are required for interfacing with a primesense carmine or xbox kinect, respectively. If you are planning on using these devices you should install those packages, otherwise they are not required.

shinshiner, 2017-07-06:
I am sure I re-source the setup.bash file. I guess there is no error in catkin_make, here are the outputs of  command `catkin_make` for autolab_core:
```
Base path: /home/shin/catkin_ws
Source space: /home/shin/catkin_ws/src
Build space: /home/shin/catkin_ws/build
Devel space: /home/shin/catkin_ws/devel
Install space: /home/shin/catkin_ws/install
####
#### Running command: "make cmake_check_build_system" in "/home/shin/catkin_ws/build"
####
-- Using CATKIN_DEVEL_PREFIX: /home/shin/catkin_ws/devel
-- Using CMAKE_PREFIX_PATH: /home/shin/catkin_ws1/devel;/opt/ros/kinetic
-- This workspace overlays: /home/shin/catkin_ws1/devel;/opt/ros/kinetic
-- Using PYTHON_EXECUTABLE: /usr/bin/python
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/shin/catkin_ws/build/test_results
-- Found gtest sources under '/usr/src/gtest': gtests will be built
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.7.6
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- autolab_core: 0 messages, 2 services
-- Configuring done
-- Generating done
-- Build files have been written to: /home/shin/catkin_ws/build
####
#### Running command: "make -j4 -l4" in "/home/shin/catkin_ws/build"
####
[  0%] Built target std_msgs_generate_messages_py
[  0%] Built target std_msgs_generate_messages_eus
[  0%] Built target std_msgs_generate_messages_nodejs
[  0%] Built target std_msgs_generate_messages_lisp
[  0%] Built target std_msgs_generate_messages_cpp
[  0%] Built target _autolab_core_generate_messages_check_deps_RigidTransformPublisher
[  0%] Built target _autolab_core_generate_messages_check_deps_RigidTransformListener
[ 16%] Built target autolab_core_generate_messages_cpp
[ 41%] Built target autolab_core_generate_messages_py
[ 58%] Built target autolab_core_generate_messages_nodejs
[ 83%] Built target autolab_core_generate_messages_eus
[100%] Built target autolab_core_generate_messages_lisp
[100%] Built target autolab_core_generate_messages
```

jeffmahler, 2017-07-06:
The output of catkin_make looks reasonable.

This error is caused by a failure to import autolab_core.srv on Line 28 of rigid_transformations.py. For me the service files are in /path/to/catkin_ws/devel/lib/python2.7/dist-packages/autolab_core/srv/.

I suspect this is a problem with sourcing in your ROS environment, or there's some compatibility issue between ROS Jade and ROS kinetic that we overlooked.

Can you respond with the following?
1) Whether or not this directory exists: /path/to/catkin_ws/devel/lib/python2.7/dist-packages/autolab_core.
2) Whether or not this directory exists: /path/to/catkin_ws/devel/lib/python2.7/dist-packages/autolab_core/srv.
3) The output of executing "from autolab_core.srv import *" in Python.

shinshiner, 2017-07-06:
The two directories do exist, but the output of `from autolab_core.srv import *` is like this:
```
WARNING:root:autolab_core not installed as catkin package, RigidTransform ros methods will be unavailable
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: No module named srv
```

I install ros follow [this instruction](http://wiki.ros.org/kinetic/Installation/Ubuntu) and I just install the base part, namely ROS-Base, because I can not install the full version (have unmet dependences).

jeffmahler, 2017-07-06:
Ok. It seems to me that your Python sys.path does not contain the directory with the srv files. Here's my next suggestions:
1) Double check the ROS environment setup:
`cd /path/to/catkin_ws
catkin_make
source devel/setup.bash`
AND try the import.
2) Check for /path/to/catkin_ws/devel/lib/python2.7/dist-packages/autolab_core/srv/__init__.py. If it does not exist then the package didn't build properly.
3) If it does exist then something is wrong with your Python sys.path. I recommend deleting the build and devel folders of the catkin_ws and re-building per [this post](http://answers.ros.org/question/52790/no-module-named-srv/).

If you confirm that the services build with the __init__.py file then the problem has to do with your Python / ROS configuration, not the autolab_core package and we won't be able to further debug the problem.

shinshiner, 2017-07-10:
I search related solutions for some days, and most of the similar problems are due to the name conflict, such as this [post](http://answers.ros.org/question/173507/cannot-import-custom-service/). But everything must have gone well in your environment, will this case happen?

shinshiner, 2017-07-10:
By the way, I have some questions about the instructions like 
```
$ cd {PATH_TO_YOUR_CATKIN_WORKSPACE}/src
$ git clone https://github.com/BerkeleyAutomation/gqcnn.git
```
If I execute these commands, the file strcture will become:
```
/home/xxx/catkin_ws/src/gqcnn/(contents in gqcnn repository)
/home/xxx/catkin_ws/src/autolab_core/(contents in autolab_core repository)
/home/xxx/catkin_ws/src/perception/(contents in percetion repository)
```

Then if I run `catkin_make`, which package am I making? Or I should copy the CMakeLists.txt of each repository into the directory `/home/xxx/catkin_ws/src/` before I execute `catkin_make`?

jeffmahler, 2017-07-10:
> I search related solutions for some days, and most of the similar problems are due to the name conflict, such as this post. But everything must have gone well in your environment, will this case happen?

I have not seen the error you are describing. Based on your first post I doubt that it is a naming issue as described in that post because the executable is being run from /home/xxx/catkin_ws/src/gqcnn/.

`Then if I run catkin_make, which package am I making? Or I should copy the CMakeLists.txt of each repository into the directory /home/xxx/catkin_ws/src/ before I execute catkin_make?`

If you are not already familiar with ROS then we highly recommend that you install t[he Python-only version](https://berkeleyautomation.github.io/gqcnn/install/install.html#python-installation). **ROS integration is not necessary for training**.

To answer your question, you are building all of the packages. We do not have the support capabilities to answer questions about ROS that are publicly available. I suggest you familiarize yourself with catkin through [the ROS catkin tutorials](http://wiki.ros.org/catkin/Tutorials). In particular see [tutorial 3](http://wiki.ros.org/catkin/Tutorials/using_a_workspace).

shinshiner, 2017-07-10:
Thank you very much! I will read these ROS materials in detail. 
There exists the last question: I first finish the python installation and I can run the `tools/training.py` in `/home/xxx/gqcnn/`. 
Then I follow the instructions to do ROS installation. After that, I have the directory `/home/xxx/catkin_ws/src/gqcnn/contents in gqcnn`. 
If I want to train the network, I should run which `training.py`? Or both are OK?

jeffmahler, 2017-07-16:
Hi, very very sorry for the delay. Yes, please run
`cd /path/to/your/gqcnn`
`python tools/training.py`

closed 2017-07-16 by jeffmahler

jaspereb, 2018-01-15:
Just for future reference.

I installed the python version (via pip) of autolab_core to test it and later installed the ROS version but forgot to remove the python version. This also results in the above error as python searches for the srv module within the autolab_core pip install, so check the path of the autolab_core module.

Removing the python install with sudo -H pip uninstall autolab_core fixes the issue, otherwise you would need to rename the ROS install module.

jeffmahler, 2018-01-15:
Thanks for the info!

kundan09, 2018-11-27:
@jeffmahler 

Please look into the issue 
**WARNING:root:autolab_perception is not installed as a catkin package -**
WARNING:root:autolab_perception is not installed as a catkin package - ROS msg conversions will not be available for image wrappers
WARNING:root:Unable to import pylibfreenect2. Python-only Kinect driver may not work properly.
WARNING:root:Failed to import ROS in Kinect2_sensor.py. Kinect will not be able to be used in bridged mode
WARNING:root:Unable to import openni2 driver. Python-only Primesense driver may not work properly
WARNING:root:Failed to import ROS in ensenso_sensor.py. ROS functionality not available
WARNING:root:Failed to import ROS in phoxi_sensor.py. PhoXiSensor functionality unavailable.
WARNING:root:Failed to import mayavi
WARNING:root:Failed to import mayavi
WARNING:root:Failed to import mayavi
WARNING:root:Failed to import mayavi
WARNING:root:Failed to import mayavi
WARNING:root:Failed to import OpenRAVE
WARNING:root:Failed to import mayavi
WARNING:root:Failed to import OpenRAVE
WARNING:dexnet.api:Failed to import DexNetVisualizer3D, visualization methods will be unavailable
test_illegal_create (test.database_test.Hdf5DatabaseTest) ... ok
test_new_database_and_graspable (test.database_test.Hdf5DatabaseTest) ... Input mesh  /home/robo/catkin_ws/src/dex-net/data/test/models/bar_clamp.obj
output mesh  /home/robo/catkin_ws/src/dex-net/.dexnet/bar_clamp_proc.obj
Loading Plugins:
Current Plugins Dir is: /usr/lib/meshlab/plugins 
Error in XMLFile: filter_measure.xml - line: -1, column: -1 - 
Error in XMLFile: filter_mutualinfo.xml - line: -1, column: -1 - 
Total 227 filtering actions
Total 13 io plugins
Opening a file with extention obj
Mesh bar_clamp.obj loaded has 1428 vn 476 fn
No Script to apply.
Mesh /home/robo/catkin_ws/src/dex-net/data/test/models/bar_clamp.obj saved as /home/robo/catkin_ws/src/dex-net/.dexnet/bar_clamp_proc.obj (1428 vn 476 fn)
Reading data.
Warning: 9 lines were ignored since they did not contain faces or vertices.
Read in 1428 vertices and 476 faces.
Resolution: 0.000716689 with real dimension 0.064502
Center of grid 0 0 0
Bound box size: (-0.0358344 -0.0358344 -0.0358344) to (0.021729 0.02 0.032251) with dimensions 100 100 100.
Computing signed distance field.
Writing results to: .dexnet/bar_clamp_proc.sdf
Processing complete.
WARNING:root:No rendered images of type segmask for stable pose pose_0
WARNING:root:No rendered images of type depth for stable pose pose_0
WARNING:root:No rendered images of type scaled_depth for stable pose pose_0
ok
test_antipodal_grasp_sampler (test.grasping_test.GraspTest) ... ok
test_contacts (test.grasping_test.GraspTest) ... ok
test_find_contacts (test.grasping_test.GraspTest) ... ok
test_force_closure (test.grasping_test.GraspTest) ... ok
test_grasp_quality_functions (test.grasping_test.GraspTest) ... ok
test_init_grasp (test.grasping_test.GraspTest) ... ok
test_init_graspable (test.grasping_test.GraspTest) ... ok
test_init_gripper (test.grasping_test.GraspTest) ... ok
test_min_norm_vector_in_facet (test.grasping_test.GraspTest) ... ok
test_wrench_in_positive_span (test.grasping_test.GraspTest) ... ok
test_gaussian_uniform_alloc (test.learning_test.LearningTest) ... ok
test_thompson_sampling (test.learning_test.LearningTest) ... ok
test_uniform_alloc (test.learning_test.LearningTest) ... ok

----------------------------------------------------------------------
Ran 15 tests in 125.268s

OK

jeffmahler, 2018-11-28:
@kundan09 I believe the autolab_perception issue may be because the package was accidentally installed through pip. Can you try:
```
pip uninstall autolab_perception
```

I am not sure about the second issue. Can you try the following and post any errors?
```
python
>>> import meshrender
>>> import pyglet
```