BerkeleyAutomation/gqcnn #7: The demo trainnig doesn't converge :(

sunbuny, 2017-07-15:
Hi～
I  am running your wiki's trainnig.py demo. But i found the trainning doesn't converge. It is so sad .
I am new to the tensorflow.
[image]

visatish, 2017-07-15:
Could you also post the output of Tensorboard here? Also, did you make any hyper-parameter modifications in the yaml configuration files? Also, could you specify which configuration yaml file and which dataset are being trained.

sunbuny, 2017-07-16:
sorry.
Here is the training.yaml. I changed the dataset_dir and output_dir and model_dir. That is all.
[image]
I dont know how to change the hyper-parameter......
The yaml is traning.yaml and the data set is adv_synth.zip

sunbuny, 2017-07-16:
[events.out.tfevents.1500121517.sun-PC.zip](https://github.com/BerkeleyAutomation/gqcnn/files/1150856/events.out.tfevents.1500121517.sun-PC.zip)

jeffmahler, 2017-07-16:
There were some erroneous parameters in the cfg/tools/training.yaml on master branch. Please do a pull from master and try again. Sorry about that.

`cd /path/to/your/gqcnn`
`git pull origin master`

closed 2017-07-16 by jeffmahler

jeffmahler, 2017-07-16:
Sorry one more parameter bug: dropout of the fully connected layers should be turned off. In general we recommend that you pull from master often as the package is still in beta mode and we are pushing updates often.

sunbuny, 2017-07-17:
OK，thanks for your kindly reply.