BerkeleyAutomation/gqcnn #8: ImportError: No module named autolab_core

TrinhTUHH, 2017-07-16:
Hi,
I'm trying to install ROS Installation by following https://berkeleyautomation.github.io/gqcnn/install/install.html#quick-start-guide. However, after cloning and running catkin_make I didn't see "autolab_core" folder in path/to/catkin_ws/devel/lib/python2.7/dist-packages/ . That's why when I run the command `python examples/policy.py` the error `ImportError: No module named autolab_core` always appears. I'm using Ubuntu 14.04, Trusty and ROS Indigo. Do you know what's the problem is?

Output of `catkin_make`:
`Base path: /home/tu/catkin_ws
Source space: /home/tu/catkin_ws/src
Build space: /home/tu/catkin_ws/build
Devel space: /home/tu/catkin_ws/devel
Install space: /home/tu/catkin_ws/install
####
#### Running command: "cmake /home/tu/catkin_ws/src -DCATKIN_DEVEL_PREFIX=/home/tu/catkin_ws/devel -DCMAKE_INSTALL_PREFIX=/home/tu/catkin_ws/install -G Unix Makefiles" in "/home/tu/catkin_ws/build"
####
-- Using CATKIN_DEVEL_PREFIX: /home/tu/catkin_ws/devel
-- Using CMAKE_PREFIX_PATH: /home/tu/catkin_ws/devel;/opt/ros/indigo
-- This workspace overlays: /home/tu/catkin_ws/devel;/opt/ros/indigo
-- Using PYTHON_EXECUTABLE: /usr/bin/python
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/tu/catkin_ws/build/test_results
-- Found gtest sources under '/usr/src/gtest': gtests will be built
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.6.19
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 1 packages in topological order:
-- ~~  - gqcnn
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin package: 'gqcnn'
-- ==> add_subdirectory(gqcnn)
-- Using these message generators: gencpp;genlisp;genpy
-- gqcnn: 2 messages, 1 services
-- Configuring done
-- Generating done
-- Build files have been written to: /home/tu/catkin_ws/build
####
#### Running command: "make -j8 -l8" in "/home/tu/catkin_ws/build"
####
Scanning dependencies of target geometry_msgs_generate_messages_cpp
Scanning dependencies of target _gqcnn_generate_messages_check_deps_GQCNNGrasp
Scanning dependencies of target _gqcnn_generate_messages_check_deps_GQCNNGraspPlanner
Scanning dependencies of target _gqcnn_generate_messages_check_deps_BoundingBox
Scanning dependencies of target sensor_msgs_generate_messages_py
Scanning dependencies of target sensor_msgs_generate_messages_lisp
Scanning dependencies of target geometry_msgs_generate_messages_py
[  0%] Scanning dependencies of target sensor_msgs_generate_messages_cpp
Built target geometry_msgs_generate_messages_cpp
[  0%] Built target sensor_msgs_generate_messages_py
[  0%] Scanning dependencies of target geometry_msgs_generate_messages_lisp
Built target geometry_msgs_generate_messages_py
[  0%] [  0%] Built target sensor_msgs_generate_messages_cpp
Built target sensor_msgs_generate_messages_lisp
[  0%] Built target geometry_msgs_generate_messages_lisp
[  0%] Built target _gqcnn_generate_messages_check_deps_BoundingBox
[  0%] [  0%] Built target _gqcnn_generate_messages_check_deps_GQCNNGrasp
Built target _gqcnn_generate_messages_check_deps_GQCNNGraspPlanner
Scanning dependencies of target gqcnn_generate_messages_py
Scanning dependencies of target gqcnn_generate_messages_lisp
Scanning dependencies of target gqcnn_generate_messages_cpp
[  9%] [ 18%] [ 27%] [ 45%] [ 45%] [ 54%] [ 63%] Generating C++ code from gqcnn/GQCNNGrasp.msg
[ 72%] Generating C++ code from gqcnn/GQCNNGraspPlanner.srv
Generating Python from MSG gqcnn/GQCNNGrasp
Generating C++ code from gqcnn/BoundingBox.msg
Generating Python from MSG gqcnn/BoundingBox
Generating Lisp code from gqcnn/GQCNNGrasp.msg
Generating Lisp code from gqcnn/BoundingBox.msg
Generating Lisp code from gqcnn/GQCNNGraspPlanner.srv
[ 81%] Generating Python code from SRV gqcnn/GQCNNGraspPlanner
[ 81%] Built target gqcnn_generate_messages_lisp
[ 90%] [100%] Generating Python msg __init__.py for gqcnn
Generating Python srv __init__.py for gqcnn
[100%] Built target gqcnn_generate_messages_py
[100%] Built target gqcnn_generate_messages_cpp
Scanning dependencies of target gqcnn_generate_messages
[100%] Built target gqcnn_generate_messages
`

jeffmahler, 2017-07-16:
Did you download and install autolab_core in your catkin workspace source folder? I don't see it from your catkin_make command.

If so, did you do: "source devel/setup.bash" from /path/to/your/catkin_ws?

TrinhTUHH, 2017-07-16:
Hi Jeff,
Thank you for your fast response! You're right that I forgot to install autolab_core. After installing autolab_core, perception, and also re-source setup.bash, I ran the command again. Now the error is:
```
Unable to import DataStreamSyncer and Recorder! Likely due to missing multiprocess
WARNING:root:Unable to import pylibfreenect2. Python-only Kinect driver may not work properly.
WARNING:root:Unable to import openni2 driver. Python-only Primesense driver may not work properly
WARNING:root:primesense_sensor.py not installed as catkin package. ROS functionality not available.
Traceback (most recent call last):
  File "examples/policy.py", line 143, in <module>
    policy = CrossEntropyAntipodalGraspingPolicy(policy_config)
  File "/home/tu/catkin_ws/src/gqcnn/./gqcnn/policy.py", line 377, in __init__
    GraspingPolicy.__init__(self, config)
  File "/home/tu/catkin_ws/src/gqcnn/./gqcnn/policy.py", line 104, in __init__
    self._gqcnn = GQCNN.load(self._gqcnn_model_dir)
  File "/home/tu/catkin_ws/src/gqcnn/./gqcnn/neural_networks.py", line 71, in load
    with open(config_file) as data_file:    
IOError: [Errno 2] No such file or directory: '/home/user/data/models/GQ-Image-Wise/config.json'
INFO:rospy.core:signal_shutdown [atexit]
tu@tu:~/catkin_ws/src/gqcnn$ python examples/policy.py --config_filename cfg/examples/policy.yaml
Unable to import DataStreamSyncer and Recorder! Likely due to missing multiprocess
WARNING:root:Unable to import pylibfreenect2. Python-only Kinect driver may not work properly.
WARNING:root:Unable to import openni2 driver. Python-only Primesense driver may not work properly
WARNING:root:primesense_sensor.py not installed as catkin package. ROS functionality not available.
Traceback (most recent call last):
  File "examples/policy.py", line 143, in <module>
    policy = CrossEntropyAntipodalGraspingPolicy(policy_config)
  File "/home/tu/catkin_ws/src/gqcnn/./gqcnn/policy.py", line 377, in __init__
    GraspingPolicy.__init__(self, config)
  File "/home/tu/catkin_ws/src/gqcnn/./gqcnn/policy.py", line 104, in __init__
    self._gqcnn = GQCNN.load(self._gqcnn_model_dir)
  File "/home/tu/catkin_ws/src/gqcnn/./gqcnn/neural_networks.py", line 71, in load
    with open(config_file) as data_file:    
IOError: [Errno 2] No such file or directory: '/home/user/data/models/GQ-Image-Wise/config.json'
INFO:rospy.core:signal_shutdown [atexit]
```

jeffmahler, 2017-07-17:
Hi, it looks like you did not download the example pretrained GQ-CNN weights.

Per [the tutorial](https://berkeleyautomation.github.io/gqcnn/tutorials/tutorial.html#grasp-planning), you must change the following parameters in cfg/examples.policy.yaml to point to a valid pre-trained GQ-CNN:
```
policy:
  gqcnn_model: /path/to/your/model/
```

You can do this by downloading the folder GQ-Image-Wise from [the pretrained weights data repo](http://bit.ly/2tAFMko) to a folder on your machine, then updating the gqcnn_model parameter in the YAML file.

Apologies that the instructions are not more clear about this. I am on vacation but I will update the docs when I get a chance.

TrinhTUHH, 2017-07-17:
It works now. Thank you very much and enjoy your holiday!

closed 2017-07-17 by TrinhTUHH