BerkeleyAutomation/gqcnn #9: How to use the dataset?

shinshiner, 2017-07-21:
Hello, I download the newest codes, but I find the result does not converge and the validation error has not changed. I doubt that I did not use the dataset properly. I use the dataset adv_synth.zip. I just extract it and set corresponding data_dir in `cfg/tools/training.yaml`, then I run `python tools/training.py`. Is it right? And I find that the large dataset is two unformatted files, how can I use them?

jeffmahler, 2017-07-27:
Hi, sorry for the delay!

With respect to training, we run a cron job every few days to check that the training tutorial converges with the current master branch (after setting `dataset_dir` and `output_dir` in training.yaml for our system). The latest job ran last night and passed.

Here is a plot of the training convergence:
[image]

Please make sure that you are up-to-date with the master branch with no local changes (e.g. `git status` should only show that `cfg/tools/training.yaml` is modified). Once you are sure that you are up-to-date with the master branch, can you please email us your training.yaml file? Please send to vsatish [at] berkeley [dot] edu.

With respect to the large dataset, please see the [README](https://berkeley.box.com/s/mrpspjwk2rxhvorard4at9fdlf637hs7) file in the [dex_net_large dataset folder](https://berkeley.box.com/s/pub2x8mtwhrzppr11nee0q6hcx0rm32w) and follow the instructions to join the two partial files.

jeffmahler, 2017-11-08:
Closing as convergence has been replicated a number of times and this not longer appears to be an issue.

closed 2017-11-08 by jeffmahler