BerkeleyAutomation/gqcnn #11: Error policy example

RobotNr5, 2017-08-18:
Hi,
after installing all the dependencies I try to get the policy example running, but there's this error:
`uburobot@uburobot:~/catkin_ws/src/gqcnn$ python examples/policy.py 
WARNING:root:autolab_core not installed as catkin package, RigidTransform ros methods will be unavailable
/usr/lib/python2.7/dist-packages/pkg_resources.py:1031: UserWarning: /home/uburobot/.python-eggs is writable by group/others and vulnerable to attack when used with get_resource_filename. Consider a more secure location (set with .set_extraction_path or the PYTHON_EGG_CACHE environment variable).
  warnings.warn(msg, UserWarning)
WARNING:root:Unable to import pylibfreenect2. Python-only Kinect driver may not work properly.
WARNING:root:Unable to import openni2 driver. Python-only Primesense driver may not work properly
WARNING:root:primesense_sensor.py not installed as catkin package. ROS functionality not available.
Traceback (most recent call last):
  File "examples/policy.py", line 106, in <module>
    from gqcnn import CrossEntropyAntipodalGraspingPolicy, RgbdImageState
ImportError: No module named gqcnn
`
I downloaded the GQ (from RSS 2017 paper) and changed the policy/gqcnn_model parameter in the policy.yaml config file to the location of the model
`# gqcnn params
  gqcnn_model: ~/Downloads/GQ_Models`

I also tried to start the example by adding _--config_filename ./cfg/examples/policy.yaml_ behind the command.

visatish, 2017-08-18:
Warning #1: This warning is being raised in rigid_transformations.py in the autolab_core module on line 24. It means that the ros srv files could not be imported. This indicates something might have gone wrong with the ros installation. Could you attach the output of your catkin_make?

Warning #2/#3: These warnings concern python libraries for the specific depth sensor used. The first is if you are using a Kinect v2. The second is the openni2 driver for the primesense carmine. Please download the appropriate library for the sensor you are using.

Warning #4: This is similar to warning #1 in that once again the ros srv files for the perception module cannot be imported.

Error: This means that the gqcnn module cannot be imported. Did you run source setup/devel.bash before running this? If that doesn't work, can you run the python shell and try importing the gqcnn module and post the output?

As a warning, we have only tested this on ros jade and people have had similar problems when running it on indigo. We are looking into this but if you could use jade that would be highly preferred and might solve the issues you are having.

RobotNr5, 2017-08-21:
#1: What should I catkin_make? Sry but you have to be more specific, I am a beginner with Ubuntu. If you mean the part for the ROS installation: How do I access the catkin output when I did this?
#2: I installed libfreenect2 and pylibfreenect2 for my Kinect2, the examples work. Do I have to copy them somewhere? Because I have seen something like this about openni with primesense.

I did run source setup/devel.bash after the catkin_make as stated in the guide. [Guide](https://berkeleyautomation.github.io/gqcnn/install/install.html#ros-installation)

> If that doesn't work, can you run the python shell and try importing the gqcnn module
How do I do that?

Unfortunately I can't switch to ROS Jade, because the robot has to run on Indigo.

visatish, 2017-08-21:
Hi,

Sorry that I wasn't too clear. You are correct in that I meant the catkin_make that is run in your catkin workspace to install the catkin packages. If you could copy and past the output displayed in the terminal after catkin_make is run that would be great. I just want to make sure no errors are being thrown.
You should not have to copy libfreenect2 and pylibfreenect2 anywhere. As far as our setup went, we did not have to copy openni2 anywhere to get it working.
Regarding the python shell, I meant the python interpreter, which can be launched by typing in python in the terminal. Then could you try 'import gqcnn' and tell us what the output is? My hunch is that it will throw an error because the module has not been properly installed.

Thank You

RobotNr5, 2017-08-22:
Hi visatish,
thank you for your help. This is the catkin_make output:
```
uburobot@uburobot:~/catkin_ws$ catkin_make
Base path: /home/uburobot/catkin_ws
Source space: /home/uburobot/catkin_ws/src
Build space: /home/uburobot/catkin_ws/build
Devel space: /home/uburobot/catkin_ws/devel
Install space: /home/uburobot/catkin_ws/install
####
#### Running command: "cmake /home/uburobot/catkin_ws/src -DCATKIN_DEVEL_PREFIX=/home/uburobot/catkin_ws/devel -DCMAKE_INSTALL_PREFIX=/home/uburobot/catkin_ws/install -G Unix Makefiles" in "/home/uburobot/catkin_ws/build"
####
-- Using CATKIN_DEVEL_PREFIX: /home/uburobot/catkin_ws/devel
-- Using CMAKE_PREFIX_PATH: /opt/ros/indigo
-- This workspace overlays: /opt/ros/indigo
-- Using PYTHON_EXECUTABLE: /usr/bin/python
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/uburobot/catkin_ws/build/test_results
-- Found gtest sources under '/usr/src/gtest': gtests will be built
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.6.19
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 7 packages in topological order:
-- ~~  - iai_kinect2 (metapackage)
-- ~~  - autolab_core
-- ~~  - gqcnn
-- ~~  - kinect2_registration
-- ~~  - kinect2_bridge
-- ~~  - kinect2_calibration
-- ~~  - kinect2_viewer
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin metapackage: 'iai_kinect2'
-- ==> add_subdirectory(iai_kinect2/iai_kinect2)
-- +++ processing catkin package: 'autolab_core'
-- ==> add_subdirectory(autolab_core)
-- Using these message generators: gencpp;genlisp;genpy
-- autolab_core: 0 messages, 2 services
-- +++ processing catkin package: 'gqcnn'
-- ==> add_subdirectory(gqcnn)
-- Using these message generators: gencpp;genlisp;genpy
-- gqcnn: 2 messages, 1 services
-- +++ processing catkin package: 'kinect2_registration'
-- ==> add_subdirectory(iai_kinect2/kinect2_registration)
-- Found OpenCL: /usr/lib/x86_64-linux-gnu/libOpenCL.so  
-- CPU based depth registration enabled
-- OpenCL based depth registration enabled
-- Performing Test OpenCL_C_WORKS
-- Performing Test OpenCL_C_WORKS - Success
-- Performing Test OpenCL_CXX_WORKS
-- Performing Test OpenCL_CXX_WORKS - Success
-- +++ processing catkin package: 'kinect2_bridge'
-- ==> add_subdirectory(iai_kinect2/kinect2_bridge)
-- Using these message generators: gencpp;genlisp;genpy
-- +++ processing catkin package: 'kinect2_calibration'
-- ==> add_subdirectory(iai_kinect2/kinect2_calibration)
-- +++ processing catkin package: 'kinect2_viewer'
-- ==> add_subdirectory(iai_kinect2/kinect2_viewer)
-- Boost version: 1.54.0
-- Found the following Boost libraries:
--   system
--   filesystem
--   thread
--   date_time
--   iostreams
--   serialization
-- checking for module 'openni-dev'
[... 65 lines cut ...]
[100%] Built target kinect2_viewer
Linking CXX executable /home/uburobot/catkin_ws/devel/lib/kinect2_bridge/kinect2_bridge
Linking CXX shared library /home/uburobot/catkin_ws/devel/lib/libkinect2_bridge_nodelet.so
[100%] Built target kinect2_bridge_nodelet
[100%] Built target kinect2_bridge
```

Your hunch is unfortunately legitimate:
```
uburobot@uburobot:~/catkin_ws$ python
Python 2.7.6 (default, Oct 26 2016, 20:30:19) 
[GCC 4.8.4] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import gqcnn
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: No module named gqcnn
```

Maybe it's because I did the import of the model wrong. This is the part of my policy.yaml file:
```
# gqcnn params
gqcnn_model: ~/Downloads/GQ_Models/model.ckpt
```
But I also tried it just adding the folder GQ_Models without pointing to a specific file.

visatish, 2017-08-22:
Hi,

The output of your catkin_make looks good so I don't believe the issue is there. Also, the gqcnn_model parameter should point to the model folder you downloaded(the directory containing the model.ckpt and other meta/index/npy/pkl files).

Here are some things you can try to debug the gqcnn problem:
Run "python -c "import sys; print(':'.join(x for x in sys.path if x))" to get the PYTHONPATH **after** running source devel/setup.bash. Could you show the output? There should be something like this in the output: 
"devel/lib/python2.7/dist-packages" 
This would signify that the modules built with catkin_make are in the PYTHONPATH. If this isn't there then that is the reason they cannot be imported. If that is there, could you next make sure that inside devel/lib/python(version#)/dist-packages/ there is a folder for gqcnn?

Thank You

RobotNr5, 2017-08-23:
Hi,
after troubles with opencv I probably crashed my whole setup, and I don't have the time to reset everything and start over. So you don't have to bother with my issue anymore. 

I thank you very much for your time and help.

visatish, 2017-08-23:
Hi,

Really sorry to hear that things did not work out. I think if you fixed up the systems problems and spent some time working with us to tweak some of the parameters that differ between our setup and your setup, there is a very high chance you could get reasonable grasps on your setup.

We are really excited that you attempted to try the GQ-CNN on your setup as it gives us more information as to what others are trying out so we can build a more generic model. If you ever decide to try this out again and or get it to work on your setup, don't hesitate to shoot me an email at vsatish@berkeley.edu.

Thanks!

closed 2017-08-23 by visatish

bazinga-IIIIII, 2020-06-23:
I remove the build & devel folders in catkin workspace.
Then rebuild, and it worked.