BerkeleyAutomation/gqcnn #12: Cannot import gqcnn service

akash-greyorange, 2017-11-07:
Hello,
  I recently installed GQCNN ROS Service and tested it with policy.py which works perfectly for me.Now I did gqcnn.launch for launching grasp planner service for giving input of my custom image but it fails to start service and following is the log on my command line.

root@akashp-Latitude-E5570:~/catkin_ws/src/gqcnn/launch# roslaunch gqcnn.launch 
... logging to /home/akashp/.ros/log/389615c4-c382-11e7-b384-f0d5bf915032/roslaunch-akashp-Latitude-E5570-26586.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://akashp-Latitude-E5570:34359/

SUMMARY
========

PARAMETERS
 * /GQCNN_grasp_planner/config: /home/akashp/catk...
 * /rosdistro: kinetic
 * /rosversion: 1.12.7

NODES
  /
    GQCNN_grasp_planner (gqcnn/grasp_planner_node.py)

auto-starting new master
process[master]: started with pid [26597]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 389615c4-c382-11e7-b384-f0d5bf915032
process[rosout-1]: started with pid [26610]
started core service [/rosout]
process[GQCNN_grasp_planner-2]: started with pid [26614]
WARNING:root:autolab_core not installed as catkin package, RigidTransform ros methods will be unavailable
WARNING:root:Unable to import pylibfreenect2. Python-only Kinect driver may not work properly.
WARNING:root:Unable to import openni2 driver. Python-only Primesense driver may not work properly
WARNING:root:primesense_sensor.py not installed as catkin package. ROS functionality not available.
Traceback (most recent call last):
  File "/home/akashp/catkin_ws/src/gqcnn/ros_nodes/grasp_planner_node.py", line 19, in <module>
    from gqcnn.srv import GQCNNGraspPlanner
ImportError: No module named srv
[GQCNN_grasp_planner-2] process has died [pid 26614, exit code 1, cmd /home/akashp/catkin_ws/src/gqcnn/ros_nodes/grasp_planner_node.py __name:=GQCNN_grasp_planner __log:=/home/akashp/.ros/log/389615c4-c382-11e7-b384-f0d5bf915032/GQCNN_grasp_planner-2.log].
log file: /home/akashp/.ros/log/389615c4-c382-11e7-b384-f0d5bf915032/GQCNN_grasp_planner-2*.log

Kindly provide me further info to solve this problem.Also I installed autolab_core still it shows autolab_core is not installed as catkin package.I did make install in build/lib/python_2.7/dist_packages/gqcnn/ also with no success.I am using ROS Kinetic on ubuntu 16.04.

jeffmahler, 2017-11-07:
This is issue appears to be specific to ROS Kinetic but we recently upgraded to Kinetic and have not experienced the problem on our setup.

Did you [install GQ-CNN as a ROS package](https://berkeleyautomation.github.io/gqcnn/install/install.html#ros-installation)?

If so, can you check for the srv folder in /path/to/your/catkin/workspace/devel/lib/python2.7/dist-packages/gqcnn/? If it does not exist then something failed in catkin_make.

akash-greyorange, 2017-11-08:
I fixed this.Actually Python is searching in catkin_ws/src/gqcnn directory for all autogenerated python code of custom message and service which should be ideally in catkin_ws/devel/lib/python2.7/dist-packages/ .So I changed this directory by adding following lines in grasp_planner_node.py and then it worked.
import os
os.chdir('/home/akashp/catkin_ws/devel/lib/python2.7/dist-packages/')

closed 2017-11-08 by akash-greyorange

jeffmahler, 2017-11-08:
Great thanks for the update! Yes that is a problem. I believe this may happen if at some point you installed gqcnn on your machine using:
```
python setup.py install
```
which would add the path to the gqcnn source to sys.path. In our experience, catkin_make will add the catkin_ws/devel/lib version to sys.path.