BerkeleyAutomation/gqcnn #13: Grasp Sampler not sampling antipodal grasps

akash-greyorange, 2017-11-14:
Hello,
 I installed gqcnn ros service on ROS kinetic and used pretrained models doing some sample tests for Grasp planning on some of my cuboid shaped objects.But Grasp sampler seems to be doing wrong sampling or not be to sample proper antipodal points along edges of my object.Kindly have a look into images attached below.
RGBD Cropped image
[image]
Sampled antipodal points
[image]
Following is my log output.I did enabled debug logs to get more clearity. Kindly help me solve this issue.Also find grasp_sampler_node.yaml file data following the logs.

root@akashp-Latitude-E5570:~/catkin_ws# roslaunch gqcnn gqcnn.launch 
... logging to /home/akashp/.ros/log/af0cabda-c938-11e7-a093-f0d5bf915032/roslaunch-akashp-Latitude-E5570-10112.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://akashp-Latitude-E5570:46762/

SUMMARY
========

PARAMETERS
 * /GQCNN_grasp_planner/config: /home/akashp/catk...
 * /rosdistro: kinetic
 * /rosversion: 1.12.7

NODES
  /
    GQCNN_grasp_planner (gqcnn/grasp_planner_node.py)

ROS_MASTER_URI=http://localhost:11311

core service [/rosout] found
process[GQCNN_grasp_planner-1]: started with pid [10161]
WARNING:root:autolab_core not installed as catkin package, RigidTransform ros methods will be unavailable
WARNING:root:Unable to import pylibfreenect2. Python-only Kinect driver may not work properly.
WARNING:root:Unable to import openni2 driver. Python-only Primesense driver may not work properly
WARNING:root:primesense_sensor.py not installed as catkin package. ROS functionality not available.
[INFO] [1510669240.601043]: Creating Grasp Policy
2017-11-14 19:50:40.799114: W tensorflow/core/platform/cpu_feature_guard.cc:45] The TensorFlow library wasn't compiled to use SSE4.1 instructions, but these are available on your machine and could speed up CPU computations.
2017-11-14 19:50:40.799143: W tensorflow/core/platform/cpu_feature_guard.cc:45] The TensorFlow library wasn't compiled to use SSE4.2 instructions, but these are available on your machine and could speed up CPU computations.
2017-11-14 19:50:40.799157: W tensorflow/core/platform/cpu_feature_guard.cc:45] The TensorFlow library wasn't compiled to use AVX instructions, but these are available on your machine and could speed up CPU computations.
2017-11-14 19:50:40.799166: W tensorflow/core/platform/cpu_feature_guard.cc:45] The TensorFlow library wasn't compiled to use AVX2 instructions, but these are available on your machine and could speed up CPU computations.
2017-11-14 19:50:40.799174: W tensorflow/core/platform/cpu_feature_guard.cc:45] The TensorFlow library wasn't compiled to use FMA instructions, but these are available on your machine and could speed up CPU computations.
[INFO] [1510669240.990180]: Grasp Sampler Server Initialized
[INFO] [1510669241.244419]: Planning Grasp
[INFO] [1510669243.259270]: Planning Grasp
2017-11-14 19:50:43,259 - root - DEBUG - Sampling 2d candidates
2017-11-14 19:50:43,262 - root - DEBUG - Depth edge detection took 0.002 sec
2017-11-14 19:50:43,262 - root - DEBUG - Found 3322 edge pixels
2017-11-14 19:50:43,318 - root - DEBUG - Normal computation took 0.056 sec
2017-11-14 19:50:43,601 - root - DEBUG - Normal pruning 0.283 sec
/home/akashp/catkin_ws/src/gqcnn/./gqcnn/image_grasp_sampler.py:40: RuntimeWarning: invalid value encountered in arccos
  in_cone_2 = (np.arccos(n2.dot(v)) < alpha)
2017-11-14 19:50:44,108 - root - DEBUG - Sampled 200 grasps from image
2017-11-14 19:50:44,108 - root - DEBUG - Sampling grasps took 0.849 sec
2017-11-14 19:50:44,108 - root - DEBUG - Computing the seed set took 0.849 sec
2017-11-14 19:50:44,153 - root - DEBUG - Tensor conversion took 0.044 sec
2017-11-14 19:50:44,154 - root - DEBUG - CEM iter 0
2017-11-14 19:50:44,154 - root - DEBUG - Predicting file 0
2017-11-14 19:50:44,286 - root - DEBUG - Predicting file 16
2017-11-14 19:50:44,416 - root - DEBUG - Predicting file 32
2017-11-14 19:50:44,545 - root - DEBUG - Predicting file 48
2017-11-14 19:50:44,677 - root - DEBUG - Predicting file 64
2017-11-14 19:50:44,818 - root - DEBUG - Predicting file 80
2017-11-14 19:50:44,953 - root - DEBUG - Predicting file 96
2017-11-14 19:50:45,088 - root - DEBUG - Predicting file 112
2017-11-14 19:50:45,222 - root - DEBUG - Predicting file 128
2017-11-14 19:50:45,356 - root - DEBUG - Predicting file 144
2017-11-14 19:50:45,489 - root - DEBUG - Predicting file 160
2017-11-14 19:50:45,632 - root - DEBUG - Predicting file 176
2017-11-14 19:50:45,772 - root - DEBUG - Predicting file 192
2017-11-14 19:50:45,913 - root - DEBUG - Prediction took 1.759 sec
2017-11-14 19:50:52,938 - root - DEBUG - GMM fitting with 20 components took 0.038 sec
2017-11-14 19:50:52,941 - root - DEBUG - GMM sampling took 0.003 sec
2017-11-14 19:50:52,960 - root - DEBUG - Tensor conversion took 0.017 sec
2017-11-14 19:50:52,960 - root - DEBUG - CEM iter 1
2017-11-14 19:50:52,960 - root - DEBUG - Predicting file 0
2017-11-14 19:50:53,113 - root - DEBUG - Predicting file 16
2017-11-14 19:50:53,249 - root - DEBUG - Predicting file 32
2017-11-14 19:50:53,375 - root - DEBUG - Predicting file 48
2017-11-14 19:50:53,507 - root - DEBUG - Prediction took 0.547 sec
2017-11-14 19:50:56,777 - root - DEBUG - GMM fitting with 6 components took 0.004 sec
2017-11-14 19:50:56,781 - root - DEBUG - GMM sampling took 0.003 sec
2017-11-14 19:50:56,802 - root - DEBUG - Tensor conversion took 0.019 sec
2017-11-14 19:50:56,802 - root - DEBUG - CEM iter 2
2017-11-14 19:50:56,802 - root - DEBUG - Predicting file 0
2017-11-14 19:50:56,981 - root - DEBUG - Predicting file 16
2017-11-14 19:50:57,117 - root - DEBUG - Predicting file 32
2017-11-14 19:50:57,254 - root - DEBUG - Predicting file 48
2017-11-14 19:50:57,391 - root - DEBUG - Prediction took 0.589 sec
2017-11-14 19:51:03,555 - root - DEBUG - GMM fitting with 6 components took 0.022 sec
2017-11-14 19:51:03,556 - root - DEBUG - GMM sampling took 0.001 sec
2017-11-14 19:51:03,576 - root - DEBUG - Tensor conversion took 0.018 sec
2017-11-14 19:51:03,577 - root - DEBUG - Predicting file 0
2017-11-14 19:51:03,742 - root - DEBUG - Predicting file 16
2017-11-14 19:51:03,883 - root - DEBUG - Predicting file 32
2017-11-14 19:51:04,013 - root - DEBUG - Predicting file 48
2017-11-14 19:51:04,151 - root - DEBUG - Final prediction took 0.574 sec
[INFO] [1510669284.796026]: Total grasp planning time: 41.5362010002 secs.
2017-11-14 19:51:24,795 - rosout - INFO - Total grasp planning time: 41.5362010002 secs.

grasp_sampler_node.yaml

# visualization
vis:
  vis_cropped_rgbd_image: 1
  vis_uncropped_color_image: 0
  vis_uncropped_depth_image: 0

# padding for the cropped images so when they are rotated to be centered on the grasp point there will not be any gaps where there is no part of the image
width_pad: 0
height_pad: 0

# inpaint
inpaint_rescale_factor: 0.5 # amount to rescale the images before inpainting (smaller numbers == faster code) 

# policy params
policy:
  # optimization params
  num_seed_samples: 200 # number of seed samples
  num_gmm_samples: 50
  num_iters: 3
  gmm_refit_p: 0.25
  gmm_component_frac: 0.4
  gmm_reg_covar: 0.01

  # gqcnn params
  gqcnn_model: /home/akashp/Work/GQ-Image-Wise

  # general params
  deterministic: 1
  gripper_width: 0.140
  crop_height: 96
  crop_width: 96
  #logging_dir: /home/akashp/Work

  # sampling params
  sampling:

    # type
    type: antipodal_depth

    # antipodality
    friction_coef: 1.0
    depth_grad_thresh: 0.0025
    depth_grad_gaussian_sigma: 1.0
    downsample_rate: 4
    max_rejection_samples: 2500

    # distance
    max_dist_from_center: 10000000
    min_dist_from_boundary: 40
    min_grasp_dist: 2.5
    angle_dist_weight: 5.0

    # depth sampling
    depth_samples_per_grasp: 1
    depth_sample_win_height: 1
    depth_sample_win_width: 1
    min_depth_offset: 0.015
    max_depth_offset: 0.0

  # visualization
  vis:
    grasp_sampling : 0
    tf_images: 0
    grasp_candidates: 1
    elite_grasps: 0
    grasp_ranking: 0
    grasp_plan: 1
    final_grasp: 1
    k: 25

jeffmahler, 2017-11-15:
Thank you for providing such detailed information.

The most concerning thing I can tell from your logs is that the sampled grasps have zero width. This could be due to incorrect camera parameters. Can you provide the output of:
```
rostopic echo <your_camera_info_topic>
```
Another potential issue is that the code is designed to only plan grasps for objects imaged with an overhead RGB-D camera that is 50-70cm above the workspace. See [our system requirements](https://berkeleyautomation.github.io/gqcnn/#system-requirements) for more info. I am interested in debugging this either way but you may have decreased performance.

To help debug, I just pushed new code to master that saves the input and output of the GraspingPolicy. Can you do the following?

1. Pull from master.
2. Uncomment the logging_dir in "general params?"
3. Re-run the code.

Then just zip the subdirectories of the logging_dir (named policy_output_<10-char-hash>) and send them to me with the .yaml file you used to run the code. I should be able to replicate your issues on our machines and then I can give better feedback.

akash-greyorange, 2017-11-17:
Hello @jeffmahler ,
 I solved it partially but still success rate is not as it should be.I am using realsense sr300 camera whose ros driver gives depth image in unsigned int 16 format which I was not converting to float for supporting float32 format which is supported by gqcnn service so my edge map was not perfect.Also my depth image seems to be noisy with holes or irregularity on object surface which seems to be problem since my edge map gets smaller and maybe because of which sampler is not able to give me perfect antipodal grasp points.Also my setup is obviously not overhead camera,its front view camera and I have maintained distance of 50 to 70 cms between object and camera.
Attached is my current grasp policy output with some cylinderical shape object
[policy_output_gmglewgifk.tar.gz](https://github.com/BerkeleyAutomation/gqcnn/files/1481288/policy_output_gmglewgifk.tar.gz)

akash-greyorange, 2017-11-20:
@jeffmahler Any guidelines or updates on improving sampling regarding this?

JACKTOO1221, 2017-11-21:
Hi, aksh-greyorange
I wanna to test the grasp planning on my own images . The color image's format is .png but the depth image's format is .npy . I've tried to transform .png to .npy , but failed . I take notice of that the number in the .npy file is float ,is there normalization in transformation?
Can you give me some advice on the transformation?

akash-greyorange, 2017-11-21:
Hello @JACKTOO1221 ,
 Actually I am working on live feed from my ROS driver for depth image so never went into saving image and then reopening file.Yes you need depth image in float32 format as input to GQCNN service and it needs to be normalized in transformation.I am doing the same over my live depth image feed.I also failed initially while saving npy file and then reopening them for feeding to GQCNN service so switched over to live feed.

JACKTOO1221, 2017-11-21:
@akash-greyorange 
Thank you very much!
I install GQCNN with the python version instead of the ROS version , so I can't switch over to live right now . It seems that I have to try again...

jeffmahler, 2017-11-27:
@akash-greyorange Apologies for the delay, I was traveling internationally for the holiday.

Thank you for the policy output file. There are multiple problems with your current image setup:

1. The depth images need to be converted to meters with float32 dtype. Technically it will run as you have found out, but the object looks extremely far away to GQ-CNN when you give it the raw uint16 images. To the best of my knowledge, [the depth image in uint16 format contains the depth in millimeters](http://www.ros.org/reps/rep-0118.html), so you can convert to the correct size by dividing by 1000.0.

2. The camera intrinsics are wrong since you are using a realsense sr300 camera. Can you send the output of the camera_info topic for your camera?

I need to run the full output on our local setup to debug your results, and I will do that ASAP this week.

jeffmahler, 2017-11-27:
@JACKTOO1221 Our Python code stores depth images as .npy since we use depth in meters instead of the raw uint16 format from ROS. You can load our depth images using `np.load('/path/to/your/depth_image.npy')` or `perception.DepthImage.open('/path/to/your/depth_image.npy')` to load the image in our custom format, which makes preprocessing easier.

akash-greyorange, 2017-11-27:
@jeffmahler  Thanks for your feedback.
1. I did convert depth image to float32 format and also divided it by 1000.0 so as to make compatible with gqcnn service but still I am not able to get correct grasp pose.Usually I get grasp pose which is hardly 3 4 cms in Z axis and 1 cms in X axis.I have made setup in which my object is at 50 to 60 cms from camera optical frame so my assumption is grasp pose will be point on my object which should be between 50 to 60 cms in Z axis in grasp pose obtained after planning.For debugging purpose I printed max value of my depth image after conversion to float which gives me 0.68 which is correct since its farthest captured point from my camera frame which is wall behind my object.
2.I am using[ this package](https://github.com/jamessergeant/ros_realsense) which aligns depth and rgb image since there is obvious shift in original depth and rgb images.Following is the output of camera info topic after running this package :
header: 
  seq: 107
  stamp: 
    secs: 1511762567
    nsecs: 153100741
  frame_id: camera_rgb_optical_frame
height: 480
width: 640
distortion_model: Plumb Bob
D: [0.11410052329301834, 0.13607729971408844, 0.003039809176698327, 0.0023750297259539366, -0.0039654686115682125]
K: [473.6129150390625, 0.0, 310.19891357421875, 0.0, 473.6129150390625, 245.44081115722656, 0.0, 0.0, 1.0]
R: [1.0, 0.0, 0.0, 0.0, 1.0, 0.0, 0.0, 0.0, 1.0]
P: [473.6129150390625, 0.0, 310.19891357421875, 0.0, 0.0, 473.6129150390625, 245.44081115722656, 0.0, 0.0, 0.0, 1.0, 0.0]
binning_x: 0
binning_y: 0
roi: 
  x_offset: 0
  y_offset: 0
  height: 0
  width: 0
  do_rectify: False
3.Yes I did inpainting of depth image using the same function as mentioned by you but my CNN predictions start to lose its accuracy after inpainting and result also vary using multiple models which are GQ-Image-Wise , GQ-Stable-Pose-Wise and GQ-Object-Wise.I dont know exact difference between these models and what model to use exactly.

akash-greyorange, 2017-11-27:
Update : there is some problem with that depth aligned camera info since if I apply original realsense ros camera info then I get following info on camera info topic and grasp point also seems to be perfect as it shows at final step.Just I need to align depth and rgb images so that whenever I get bounding box from detector my depth image will be cropped perfectly.
I also did inpainting and without any medain blurring with GQ-Stable-Pose-Wise model and getting good results.
header: 
  seq: 66533
  stamp: 
    secs: 1511766741
    nsecs: 350297660
  frame_id: camera_depth_optical_frame
height: 480
width: 640
distortion_model: plumb_bob
D: [0.11410052329301834, 0.13607729971408844, 0.003039809176698327, 0.0023750297259539366, -0.0039654686115682125]
K: [473.6129150390625, 0.0, 310.19891357421875, 0.0, 473.6129150390625, 245.44081115722656, 0.0, 0.0, 1.0]
R: [1.0, 0.0, 0.0, 0.0, 1.0, 0.0, 0.0, 0.0, 1.0]
P: [473.6129150390625, 0.0, 310.19891357421875, 0.025700001046061516, 0.0, 473.6129150390625, 245.44081115722656, 0.0012667370028793812, 0.0, 0.0, 1.0, 0.004230356775224209]
binning_x: 0
binning_y: 0
roi: 
  x_offset: 0
  y_offset: 0
  height: 0
  width: 0
  do_rectify: False
---
Is there any api to align or project depth image from rgb frame?I also need horizontal grasp as per my setup so I am having tough time to transform vertical grasp pose to horizontal frame.Is there any transformation available for this one?

akash-greyorange, 2017-11-28:
Update : I solved the transformation from vertical to horizontal grasp. I have one outstanding issue now which is depth rgb alignment.SR300 ROS driver does give me depth registered rectified image but rectified image seems to be creating problem when fed as input to gqcnn service.So I need unrectified image or can rectified image work after some pre-processing?Or Do I need to align depth to rgb myself without using rectification?

jeffmahler, 2017-12-04:
@akash-greyorange Ok, great to hear that the grasp points look better! Sorry for the delay.

You should use the depth registered and rectified image if possible. What problems are you encountering with the rectified image?

We don't use the Realsense much in my lab but here is how to get registered images to the best of my knowledge:

1. Install the latest version of librealsense. Update if necessary. I recommend installing from source since the aptitude seems to want to make some kernel mods that aren't necessary for a recent Ubuntu 16.04 install.
2. Install the latest version of the ROS drivers: `sudo apt install ros-kinetic-realsense-camera`
3. Launch the rgbd driver: `roslaunch realsense_camera {your_camera_model}_nodelet_rgbd.launch`

Then the corresponding images are `/camera/rgb/image_rect_color` and `/camera/depth_registered/sw_registered/image_rect`. Both are in frame /camera_rgb_optical_frame. This works on my SR300.

Also, if you are running the grasp_planner_node then you should use the camera info on `/camera/depth_registered_sw_registered/camera_info`. The one you posted is for the color camera, I believe.

akash-greyorange, 2017-12-05:
Hello @jeffmahler , Yes I did tried rectified image now with hole inpainting which is working good in terms of sampling i.e sampler gets sufficient antipodal points and actually long back when I tried rectified depth image as input to gqcnn it was without hole inpainting which created perception that rectified image wont work.Now things are smooth with rectified image also.I am taking raw depth image,converting cvimage to numpy array,saving it in numpy file,again opening it with DepthImage .open api for preprocessing it as you mentioned in one of your last comment.And then I am sending ros message format of this image to gqcnn.Now issue which I am facing is related to prediction by CNN.I observed that I need to vary crop_height and crop_width parameters under general parameters in grasp_planner_node.yaml in order to get better predictions and this tuning is somewhere related to width of my object.Also I started tuning depth sampling parameters like depth_samples_per_grasp,depth_sample_win_height,depth_sample_win_width these parameters to improve the results but still max to max I get grasp quality upto 0.80 which is 80 percent.I need to improve this prediction.What these parameters actually mean and how to get optimum results out of them?
Find attached latest results of my experiments and help me to get better predictions.
[policy_output_aaytwzqrqm.tar.gz](https://github.com/BerkeleyAutomation/gqcnn/files/1530061/policy_output_aaytwzqrqm.tar.gz)
Also find my current grasp_planner_node.yaml file details
# visualization
vis:
  vis_cropped_rgbd_image: 1
  vis_uncropped_color_image: 0
  vis_uncropped_depth_image: 0

# padding for the cropped images so when they are rotated to be centered on the grasp point there will not be any gaps where there is no part of the image
width_pad: 50
height_pad: 50

# inpaint
inpaint_rescale_factor: 0.5 # amount to rescale the images before inpainting (smaller numbers == faster code) 

# policy params
policy:
  # optimization params
  num_seed_samples: 200 # number of seed samples
  num_gmm_samples: 50
  num_iters: 3
  gmm_refit_p: 0.25
  gmm_component_frac: 0.4
  gmm_reg_covar: 0.01

  # gqcnn params
  gqcnn_model: /home/akash.p/Work/GQ-Stable-Pose-Wise

  # general params
  deterministic: 1
  gripper_width: 0.140
  crop_height: 300              #Need to vary this width and height for better predictions
  crop_width: 300
  logging_dir: /home/akash.p/Work/Grasping_logged_data

  # sampling params
  sampling:

    # type
    type: antipodal_depth

    # antipodality
    friction_coef: 0.1
    depth_grad_thresh: 0.02             
    depth_grad_gaussian_sigma: 1.0
    downsample_rate: 4
    max_rejection_samples: 2500

    # distance
    max_dist_from_center: 10000000
    min_dist_from_boundary: 1
    min_grasp_dist: 2.5
    angle_dist_weight: 5.0

    # depth sampling
    depth_samples_per_grasp: 3        #Need to increase for big objects(3 for horlicks)
    depth_sample_win_height: 10       #Need to increase for big objects(keep 10 for horlicks)
    depth_sample_win_width: 10
    min_depth_offset: 0.015
    max_depth_offset: 0.0

  # visualization
  vis:
    grasp_sampling : 0
    tf_images: 0
    grasp_candidates: 0
    elite_grasps: 0
    grasp_ranking: 0
    grasp_plan: 0
    final_grasp: 1
    k: 25

akash-greyorange, 2017-12-06:
Update @jeffmahler  : I observed that GQCNN predicts grasps for object kept upto distance of 50 cms from camera frame.Beyond that I am not receiving any good grasp predictions.Is this because of trained model been trained with object kept at 50 cms from camera frame?Can this distance changed if I try my own model on my set of objects?

akash-greyorange, 2018-01-03:
@jeffmahler  Any update on this issue?Please consider my last update only.

> Update @jeffmahler : I observed that GQCNN predicts grasps for object kept upto distance of 50 cms from camera frame.Beyond that I am not receiving any good grasp predictions.Is this because of trained model been trained with object kept at 50 cms from camera frame?Can this distance be changed if I train my own model on my set of objects depth images?

jeffmahler, 2018-01-03:
Very sorry for taking so long. I just ran your policy output and looked things over. Many things to address:

> I observed that GQCNN predicts grasps for object kept upto distance of 50 cms from camera frame.Beyond that I am not receiving any good grasp predictions.Is this because of trained model been [...]

Yes, the distance is most likely the cause. On our setup we are able to operate at up to 80 cms, but because you are using a different camera and viewing angle, this could be problematic.

In theory this can be fixed by training a model on a set of images with updated parameters for your setup. However, the dataset generation code is not public at this time.

> I observed that I need to vary crop_height and crop_width parameters under general parameters in grasp_planner_node.yaml in order to get better predictions and this tuning is somewhere related to width of my object.

I noticed that your gripper_width is significantly larger than ours (14cm versus 5cm). This may be related to the issue because GQ-CNN is only trained on a 5cm gripper. Modifying the crop parameters will probably not lead to the desired result because the GQ-CNN is expecting the input to be a specific scale.

How are you defining "better predictions"? The prediction Q = 0 is not necessarily problematic in the sense that the GQ-CNN does not expect grasps to succeed on images that are very different from the ones used in training. It might just be the case that our public models are not suitable for your application. 

> lso I started tuning depth sampling parameters like depth_samples_per_grasp,depth_sample_win_height,depth_sample_win_width these parameters to improve the results but still max to max I get grasp quality upto 0.50 which is 50 percent.I need to improve this prediction.What these parameters actually mean and how to get optimum results out of them?

I don't think you should have to modify these parameters to get gqcnn to work, but for more info: 

- _depth_samples_per_grasp_: How many samples of depth to take for each antipodal grasp in image space. The first step in the algorithm is to sample 2D antipodal grasps using edge detection on the depth image and then to create multiple 3D grasp hypotheses from each by sampling depth values
- _depth_sample_win_height_, _depth_sample_win_width_: The dimensions of a window used to determine a lower bound on the depth for grasp sampling

> Find attached latest results of my experiments and help me to get better predictions.
policy_output_aaytwzqrqm.tar.gz

The biggest problem I noticed when running this is that the object boundaries are very near the boundaries of the image. This can cause the transformed input images to the GQ-CNN to contain zeros outside of the original image boundary. See the image below.

[image]

Are you currently cropping a bounding box around the object before inputting into the GQ-CNN. If so, one fix is to use a segmask array that is 255 inside of a bounding box around the object and zero elsewhere instead of cropping the image. You can pass such a segmask into the policy as follows:
```
binary_im = perception.BinaryImage(segmask)
state = RgbdImageState(rgbd_im, camera_intr, segmask=binary_im)
```

akash-greyorange, 2018-01-04:
Hello @jeffmahler , Thanks for responding.
> Yes, the distance is most likely the cause. On our setup we are able to operate at up to 80 cms, but because you are using a different camera and viewing angle, this could be problematic.
>
> In theory [...]
So this tells me I will need to train GQCNN with my objects and gripper width for getting perfect results.Can you give access to dataset generation code?This will be really helpful for me to take things further.

> Are you currently cropping a bounding box around the object before inputting into the GQ-CNN. If so, one fix is to use a segmask array that is 255 inside of a bounding box around the object and zero elsewhere instead of cropping the image.
Ok will try this one.Also I am cropping bounding box from top and bottom so that GQCNN wont give me any grasp wihich will hold object from top since I am dealing with horizontal grasping system.I will take care of maintaining object boundaries well within bounding box so that tf image wont be affected.Now only thing which can take things up for me is my own model with my gripper width.I went through training.py which can create model for me but I will need to generate dataset first.So please help me further regarding this.

jeffmahler, 2018-01-08:
@akash-greyorange Yes in theory you could get better results with dataset generation code.

We are working on a completely revised version of dex-net and gqcnn that supports data generation because we feel that this will be better for the projects long-term. We'd prefer to support that code when it's ready rather than release a half-baked version. We didn't share the previous one because the code was somewhat inflexible and raised many additional installation issues that we were not willing to support. The new version will be ready sometime around April or May.

Please keep in mind that your use case with the side camera view is outside the scope of what we have tested in research up to this point. If I get some time in the next few weeks then I will post the old script for dataset generation and let you know, but I will not have time to support it.

akash-greyorange, 2018-01-09:
> We are working on a completely revised version of dex-net and gqcnn that supports data generation because we feel that this will be better for the projects long-term. We'd prefer to support that [...]

That looks great.It will be really helpful for long term project development.

> Please keep in mind that your use case with the side camera view is outside the scope of what we have tested in research up to this point. If I get some time in the next few weeks then I will post the old script for dataset generation and let you know, but I will not have time to support it.

I am completely aware of this thing and infact I got good results if object is at 50 cms from camera frame.I actually grabed many novel objects like boxes and bottles which gave me some confidence.Please do share old script for dataset generation code.I will look into it and regarding installation support I will fix it on my own if possible.

akash-greyorange, 2018-01-09:
One more question.Not related to current issue.
I recently came across Dex net 3.0 paper. Awesome work :).When are you releasing code for Dex Net 3.0 I mean ROS service?This part is also helpful for our project development since we are working both on suction as well as fingered grippers.

jeffmahler, 2018-01-16:
@akash-greyorange Sounds great, I will try to release some data generation code soon.

For Dex-Net 3.0, the ROS service is in progress. You can look at the "dev_jeff" branch of gqcnn for a sneak peek but I can't support that quite yet (the ROS service doesn't even exist but we just need to make a few message types). Basically it will be available as soon as I have time to clean up this dev branch and merge it into master

akash-greyorange, 2018-01-25:
@jeffmahler  Okay Thanks for the update. Waiting to hear new updates from you on this.

jeffmahler, 2018-01-30:
@akash-greyorange I've just pushed the script to master in the dex-net repo at tools/generate_gqcnn_dataset.py. You can only generate datasets for existing objects but you can change the camera and gripper. Be sure to update the following repos before use:
- meshpy
- visualization

akash-greyorange, 2018-01-30:
@jeffmahler  Many Thanks for the update.I will try this out.What about new objects?How can that be possible?I may have requirement for generating dataset on objects present in my inventory.

jeffmahler, 2018-01-30:
You can create a database using the custom-databases branch of dex-net and then the dataset generation script can use it. See [this tutorial](https://docs.google.com/document/d/1oo3PKV8we3qSwbyGvCbCCIY3EAhSEqtRY_CjoodA7eo/edit?usp=sharing) for more info on generating grasps.

akash-greyorange, 2018-01-30:
Thanks.Access is restricted to custom-databases branch as well as tutorial in link above.Kindly provide me access.

jeffmahler, 2018-01-30:
Sorry the tutorial had the wrong permissions but [the branch is public](https://github.com/BerkeleyAutomation/dex-net/tree/custom-databases). I just accessed it without providing github credentials.

akash-greyorange, 2018-01-30:
@jeffmahler Thanks alot. I will work on this training related things and will update you incase of any query. Thanks alot for wonderful support. Closing this issue since all queries are resolved. :)

closed 2018-01-30 by akash-greyorange