BerkeleyAutomation/gqcnn #14: autolab_core not installed as catkin package

Cdfghglz, 2017-12-04:
Hi, I tried to follow the installation instructions for ROS install. My setup is U16.04 with ROS Kinetic.

For making autolab_core and perception I get:
```
Base path: /home/admin/catkin_ws
Source space: /home/admin/catkin_ws/src
Build space: /home/admin/catkin_ws/build
Devel space: /home/admin/catkin_ws/devel
Install space: /home/admin/catkin_ws/install
####
#### Running command: "make cmake_check_build_system" in "/home/admin/catkin_ws/build"
####
####
#### Running command: "make -j8 -l8" in "/home/admin/catkin_ws/build/autolab_core"
####
[  0%] Built target std_msgs_generate_messages_lisp
[  0%] Built target std_msgs_generate_messages_nodejs
[  0%] Built target std_msgs_generate_messages_py
[  0%] Built target std_msgs_generate_messages_cpp
[  0%] Built target std_msgs_generate_messages_eus
[  0%] Built target _autolab_core_generate_messages_check_deps_RigidTransformListener
[  0%] Built target _autolab_core_generate_messages_check_deps_RigidTransformPublisher
[  9%] Built target autolab_core_generate_messages_cpp
[ 36%] Built target autolab_core_generate_messages_py
[ 63%] Built target autolab_core_generate_messages_eus
[ 81%] Built target autolab_core_generate_messages_nodejs
[100%] Built target autolab_core_generate_messages_lisp
[100%] Built target autolab_core_generate_messages
```
and
```
Base path: /home/admin/catkin_ws
Source space: /home/admin/catkin_ws/src
Build space: /home/admin/catkin_ws/build
Devel space: /home/admin/catkin_ws/devel
Install space: /home/admin/catkin_ws/install
####
#### Running command: "make cmake_check_build_system" in "/home/admin/catkin_ws/build"
####
####
#### Running command: "make -j8 -l8" in "/home/admin/catkin_ws/build/perception"
####
[  0%] Built target std_msgs_generate_messages_cpp
[  0%] Built target std_msgs_generate_messages_lisp
[  0%] Built target std_msgs_generate_messages_eus
[  0%] Built target std_msgs_generate_messages_py
[  0%] Built target std_msgs_generate_messages_nodejs
[  0%] Built target _perception_generate_messages_check_deps_ImageBuffer
[ 16%] Built target perception_generate_messages_lisp
[ 50%] Built target perception_generate_messages_py
[ 66%] Built target perception_generate_messages_nodejs
[ 83%] Built target perception_generate_messages_eus
[100%] Built target perception_generate_messages_cpp
[100%] Built target perception_generate_messages
```
Also */path/to/catkin_ws/devel/lib/python2.7/dist-packages/autolab_core* and */path/to/catkin_ws/devel/lib/python2.7/dist-packages/autolab_core/srv* exist.

When after sourcing *...devel/setup.bash* I ```import autolab_core``` in python, I get
```WARNING:root:autolab_core not installed as catkin package, RigidTransform ros methods will be unavailable```.

Moreover when i  ```import gqcnn```, in addition to the warnings i get
```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/admin/catkin_ws/devel/lib/python2.7/dist-packages/gqcnn/__init__.py", line 35, in <module>
    exec(__fh.read())
  File "<string>", line 1, in <module>
  File "/home/admin/catkin_ws/src/gqcnn/./gqcnn/neural_networks.py", line 17, in <module>
    from gqcnn import InputDataMode, TrainingMode
ImportError: cannot import name InputDataMode
```
Could anybody give me some hint please?

Some more details:
- OpenCV 3.3 - ```import cv2``` is ok
- Tensorflow 1.3 - ran some examples, its ok with CUDA 9.0
- other requirements satisfied

jeffmahler, 2017-12-04:
For the autolab_core warning, did you install autolab_core using `python setup.py install` or `python setup.py develop` at any point before installing as a ROS package? This is the only way I have seen this error occur before, although it's possible.

If not, can you send the output of
```
import autolab_core
print autolab_core.__file__
```
If that doesn't print _/path/to/catkin_ws/devel/lib/python2.7/dist-packages/autolab_core_ then that's a problem.

The second error is particularly odd and I haven't observed that before. My best guess is that the failure is because InputDataMode is imported from gqcnn in neural_networks.py before it gets imported in the __init__.py file. I just pushed an update to master to avoid this. Can you try pulling and let me know if that helps?

Cdfghglz, 2017-12-05:
Hi

Thank you for the quick reply!

I most probably have used the python install command. I found and erased *./usr/local/lib/python2.7/dist-packages/autolab_core-0.1.0-py2.7.egg* which was probably shadowing the one built by catkin and now i can import autolab without warnings!

Also for the second error: after pulling, everything works well!

Thanks!

closed 2017-12-05 by Cdfghglz