BerkeleyAutomation/gqcnn #15: gqcnn with realsense

hironoriyh, 2017-12-20:
Hi, I am wondering how to setup realsense SR300. Can I test your code with a set of RGB and depth images from SR300? or should I need to change camera extrinsic/intrinsic?

jeffmahler, 2018-01-03:
Hi, you can run the code with images from an Intel Realsense but if you are executing grasps on a real robot then the hand-eye calibration may be off without using the correct intrinsics. We have used gqcnn with an Intel Realsense SR300 using [this intrinsics file](https://berkeley.box.com/s/95q0atqefxb2pmx1pl6givl499ebrzi8) but there is some variance between sensors.

Are you experiencing any specific issues?

hironoriyh, 2018-01-04:
Thanks for your answer @jeffmahler . Yes, I'd like to try grasp with a real robot. I found prime sense and kinect2 are supported in [the perception repository](https://github.com/BerkeleyAutomation/perception/blob/master/perception/rgbd_sensors.py) but not SR300. It's nice to see how you integrated SR300, although it might be more relevant to ask at the perception repo. Thanks.

jeffmahler, 2018-01-04:
That's correct. Sorry we do not yet have support in the perception repo but that should not be a huge deal if you run the [GQ-CNN grasp planning service](https://github.com/BerkeleyAutomation/gqcnn/blob/master/ros_nodes/grasp_planner_node.py). You can pass in an arbitrary color and depth image along with the camera info topic from the RealSense SR300 in a [GQCNNGraspPlanner](https://github.com/BerkeleyAutomation/gqcnn/blob/master/srv/GQCNNGraspPlanner.srv).

We are planning on adding a class to support the SR300 but I'm not sure how soon we'll push that to master.

closed 2018-01-08 by jeffmahler