BerkeleyAutomation/gqcnn #23: Error in training a GQCNN

benzben, 2018-06-12:
Hi,
trying to do a training like in the Documentation in https://berkeleyautomation.github.io/gqcnn/tutorials/tutorial.html
will result in returning an error by using

>>> with gqcnn.get_tf_graph().as_default():
...      SGDOptimizer.optimize()

Output:
INFO:root:Saving model to /home/mr/gqcnn/myWeights/model_skfcmqrwkj
Traceback (most recent call last):
  File "<stdin>", line 2, in <module>
  File "gqcnn/sgd_optimizer.py", line 154, in optimize
    self._setup()
  File "gqcnn/sgd_optimizer.py", line 981, in _setup
    self._copy_config()
  File "gqcnn/sgd_optimizer.py", line 955, in _copy_config
    shutil.copyfile(this_filename, out_train_filename)
  File "/usr/lib/python2.7/shutil.py", line 96, in copyfile
    with open(src, 'rb') as fsrc:
IOError: [Errno 2] No such file or directory: ''

Is there a way to use an already existing pretrained model so we save the time needed by this task? 
At the moment we only want display some pointclouds/grasp points as some kind of "demonstration".

mabelzhang, 2018-06-13:
What is the path for the no such file or directory error?
I'm guessing you downloaded some weights file, maybe GQ-Adv, and it doesn't run directly from there. If that's the case, most likely the config.json file in GQ-Adv is pointing to a path on their machine. You just have to edit it to point to paths of GQ-Adv on your machine.

I was able to follow the tutorial (or really, just running gqcnn/tools/training.py) to train adv_synth dataset, both from scratch, and fine-tune from the downloaded GQ-Adv weights (uncomment the corresponding section in training.py). The Analyze section of training.py also works for test-time prediction, once you manually edit the config.json file to point to paths on your machine.

jeffmahler, 2018-09-18:
Closing due to inactivity. It appears that this may have been fixed from the previous responses.

closed 2018-09-18 by jeffmahler

ZJDZJD123, 2019-03-31:
> Hi,
> trying to do a training like in the Documentation in https://berkeleyautomation.github.io/gqcnn/tutorials/tutorial.html
> will result in returning an error by using
> 
> > > > with gqcnn.get_tf_graph().as_default():
> > > > ...      SGDOptimizer.optimize()
> 
> Output:
> INFO:root:Saving model to /home/mr/gqcnn/myWeights/model_skfcmqrwkj
> Traceback (most recent call last):
> File "", line 2, in
> File "gqcnn/sgd_optimizer.py", line 154, in optimize
> self._setup()
> File "gqcnn/sgd_optimizer.py", line 981, in _setup
> self._copy_config()
> File "gqcnn/sgd_optimizer.py", line 955, in _copy_config
> shutil.copyfile(this_filename, out_train_filename)
> File "/usr/lib/python2.7/shutil.py", line 96, in copyfile
> with open(src, 'rb') as fsrc:
> IOError: [Errno 2] No such file or directory: ''
> 
> Is there a way to use an already existing pretrained model so we save the time needed by this task?
> At the moment we only want display some pointclouds/grasp points as some kind of "demonstration".

Dear  @benzben,

       How did you solve your first problem (No such file or directory: '')? I get the same problem but can't solve it by the previous responses,

        Looking forward to your kindly reply.

Sincerely,
       ZJD.