BerkeleyAutomation/gqcnn #24: Unable to close session

ankurverma-git, 2018-06-13:
**System information**
1. Ubuntu 14.04 LTS
2. VMware Player 12 (virtual machine for Ubuntu on windows 10 computer , 64 bit, 8 GB RAM)
3. Anaconda for Python 2.7
4. Tensorflow 1.8.0

**Description of the problem**

I am using the GQ-CNN to perform robust grasps on a physical robot. I have successfully trained a GQ-CNN model on the Adv-Synth dataset, with a final validation error of 1.1%. When I run the following code from the Prediction heading on the page https://berkeleyautomation.github.io/gqcnn/tutorials/tutorial.html, then an 'Attribute error' shows up that : 'NoneType' object has no attribute 'close'.

**Source code**

images = np.load('/home/aver2/Documents/extract/arr_0.npy')
poses = np.load('/home/aver2/Documents/extract/arr_02.npy')
poses = poses.reshape(4000,1)

gqcnn = GQCNN.load('/home/aver2/Documents/output_weights_final/model_nzpnizutxr')
output = gqcnn.predict(images, poses[1000:2000])
pred_p_success = output[:,1]
gqcnn.close_session()  

**Error**

AttributeError                            Traceback (most recent call last)
<ipython-input-267-bf401a4dfd15> in <module>()
      7 output = gqcnn.predict(images, poses[1000:2000])
      8 pred_p_success = output[:,1]
----> 9 gqcnn.close_session()

/home/aver2/catkin_ws/src/gqcnn/gqcnn/neural_networks.pyc in close_session(self)
    545         """ Close tensorflow session """
    546         with self._graph.as_default():
--> 547             self._sess.close()
    548             self._sess = None
    549 

AttributeError: 'NoneType' object has no attribute 'close'

Also, if I print the pred_p_success, then the values get printed and error shows up at the last. I have attached a snapshot of the error, with the printed values of pred_p_success just above it. If I remove the line - gqcnn.close_session(), then the values are printed and no error shows up.

Now, when I run the analysis module after predict module, there is a 'Not Found Error' every time. The code and error for analysis module (https://berkeleyautomation.github.io/gqcnn/tutorials/tutorial.html) are as follows:

**Code**

analysis_config = YamlConfig('/home/aver2/catkin_ws/src/gqcnn/cfg/tools/analyze_gqcnn_performance.yaml') # Sample config: 'cfg/tools/analyze_gqcnn_performance.yaml'
analyzer = GQCNNAnalyzer(analysis_config)
analyzer.analyze()

**Error**

NotFoundError: Unsuccessful TensorSliceReader constructor: Failed to find any matching files for /home/aver2/Documents/output_weights_final/model_nzpnizutxr/mini_dexnet_all_trans_01_20_17_trained_06_18_17/model.ckpt

Also, there is no model.ckpt file in my system.

Please let me know if I need to give any other details. Thanks in advance for the consideration and help!!!

[image]

visatish, 2018-06-24:
Hi,

Thanks for using the gqcnn module! Sorry for responding to this so late, the last few weeks were fairly busy. Regarding your first issue, the explicit call to gqcnn.close_session() should actually not be there, as the predict() method will automatically open and close a tensorflow session if one was not already open before the call to predict(). Thus it is trying to close a session that does not exist anymore and already has been deleted. I will push a change to this in the documentation soon. Sorry for the confusion.
Regarding your second issue, there are two things I believe are wrong with your analysis config file, which are mistakes on our part with regards to unclear documentation. 
1) 'model_dir' is a poor naming convention in this case because it does not point to the actual weight directory, but to a location containing multiple models. So in your case it would point to .../output_weights_final/. This done so that multiple models can be analyzed at once using the script.
2) You need to change the actual model name, which is now 'mini_dexnet_all_trans_01_20_17_trained_06_18_17' to that of your specific model 'model_nzpnizutxr'. Then it will append this to the 'model_dir' path when trying to load models. If you want to analyze multiple models simply add another model entry in the config file.
I will also push a change to the naming/documentation for this soon.

Please let me know if you experience any further problems!

ankurverma-git, 2018-06-25:
Thanks for the consideration!!!

I mention the following two points:

1. The predict funciton works correctly after deleting the line, 'gqcnn.close_session()'.

2. On the second issue (analysis module), my analysis config file has the following new names (after correction):
model_dir:  /home/aver2/Documents/output_weights_final
models: model_nzpnizutxr: 

When I run the code, there is an IOError:  No such file or directory: 'config.json'

**Code**
analysis_config = YamlConfig('/home/aver2/catkin_ws/src/gqcnn/cfg/tools/analyze_gqcnn_performance.yaml') # Sample config: 'cfg/tools/analyze_gqcnn_performance.yaml'
analyzer = GQCNNAnalyzer(analysis_config)
analyzer.analyze()

**Error**
IOError                                   Traceback (most recent call last)
<ipython-input-114-5ac4cb86e8d1> in <module>()
      1 analysis_config = YamlConfig('/home/aver2/catkin_ws/src/gqcnn/cfg/tools/analyze_gqcnn_performance.yaml') # Sample config: 'cfg/tools/analyze_gqcnn_performance.yaml'
      2 analyzer = GQCNNAnalyzer(analysis_config)
----> 3 analyzer.analyze()

/home/aver2/catkin_ws/src/gqcnn/gqcnn/gqcnn_analyzer.pyc in analyze(self)
     64 
     65         # run predictions
---> 66         self._run_predictions()
     67 
     68         # finally plot curves

/home/aver2/catkin_ws/src/gqcnn/gqcnn/gqcnn_analyzer.pyc in _run_predictions(self)
     98             model_subdir = os.path.join(self.model_dir, model_name)
     99             model_config_filename = os.path.join('config.json')  #('/home/aver2/Documents/output_weights_final/model_nzpnizutxr/config.json') 
--> 100             with open(model_config_filename) as data_file:
    101                     model_config = json.load(data_file)
    102             model_type = self.models[model_name]['type']

IOError: [Errno 2] No such file or directory: 'config.json'

**What I already tried/know**
1. The config.json file is present in my system. It's location is /home/aver2/Documents/output_weights_final/model_nzpnizutxr/config.json, with it's contents in the correct format. I have the permissions to read and write to the file.
2. I have run the original file without making any changes, and also ran it after changing the 'model_dir' to my local trained model's address if I trianed a model earlier. Same error in both cases.
3. I tried by specifying the absolute path in above code, as follows:
   99             model_config_filename = '/home/aver2/Documents/output_weights_final/model_nzpnizutxr/config.json' 
Still same error
4. I also ran the code using some smaller portions of the adv_synth dataset (batching- by creating new folders for datasets and models (output)), and all of them give the same error. Also, config.json file is present in all the generated model's folders.

Thanks in advance for the consideration and help!!!

visatish, 2018-06-25:
Hi @ankurverma-git,

A few questions:
1) In your above response your line 99 in '/home/aver2/catkin_ws/src/gqcnn/gqcnn/gqcnn_analyzer.pyc' seems inconsistent with that of the [master branch](https://github.com/BerkeleyAutomation/gqcnn/blob/master/gqcnn/gqcnn_analyzer.py#L99). It is missing the first model directory path that is prefixed. I'm not sure if you changed this. I'm a bit suspicious of this as you should be getting an error saying '/home/aver2/Documents/output_weights_final/model_nzpnizutxr/config.json' could not be found not 'config.json' could not be found as 'config.json' would mean it is searching in the same directory as the analysis script.

Also, can you run the following code in a python shell:

'''
import json

with open('/home/aver2/Documents/output_weights_final/model_nzpnizutxr/config.json') as data_file:
                    model_config = json.load(data_file)
'''

If the config file is where it should be and accessible, then this should not error out.

Thanks,
Vishal

ankurverma-git, 2018-06-28:
Hi @visatish,

The above error got resolved when I aligned my code with that in the master branch. Thanks!!!

Next, I downloaded the Adv-Synth dataset again and trained a new model on it. The analysis gave following results:

INFO:root:Total evaluation time: 382.807 sec
INFO:root:Beginning Plotting
INFO:root:Model model_kacjsmchuo training error rate: 0.808
INFO:root:Model model_kacjsmchuo validation error rate: 1.360

Also, the ROC curve and Precision recall curve, are both similar to that in the documentation.

Finally, when I was trying to visualize the GQCNN predictions using the following code, an 'Attribute error' occured.

**Code**
visualizer = GQCNNPredictionVisualizer(visualization_config)
visualizer.visualize()

**INFO**
INFO:root:Setting up for visualization.
INFO:root:Reading filenames
INFO:root:Loading GQ-CNN
INFO:root:Visualizing false_positive
INFO:root:Loading Image File: depth_ims_tf_table_00125.npz Pose File: hand_poses_00125.npz Label File: robust_ferrari_canny_00125.npz

**Error**
AttributeError                            Traceback (most recent call last)
<ipython-input-15-cb2433974afd> in <module>()
      1 visualizer = GQCNNPredictionVisualizer(visualization_config)
----> 2 visualizer.visualize()

/home/aver2/catkin_ws/src/gqcnn/gqcnn/gqcnn_prediction_visualizer.pyc in visualize(self)
     75             hand_poses_tensor = np.load(os.path.join(self.data_dir, pose_filename))['arr_0']
     76 
---> 77             pose_tensor = self._read_pose_data(hand_poses_tensor, self.input_data_mode)
     78 
     79             # score with neural network

/home/aver2/catkin_ws/src/gqcnn/gqcnn/gqcnn_prediction_visualizer.pyc in _read_pose_data(self, pose_arr, input_data_mode)
    267             sliced pose_data corresponding to input data mode
    268         """
--> 269         #if input_data_mode == InputDataMode.PARALLEL_JAW:
    270             #return pose_arr[:,2:3]
    271         if input_data_mode == InputDataMode.SUCTION:

AttributeError: class InputDataMode has no attribute 'PARALLEL_JAW'

**What I did/tried**
1. I ran the above code for different datasets (smaller parts of adv-synth dataset with all the necessary script files), with all giving the same error.
2. I checked for the consistency of my file with that of the master branch. 
3. I commented line 269 and 270, and changed the elif command to if command in line 271. The following error showed up:

AttributeError                            Traceback (most recent call last)
<ipython-input-3-cb2433974afd> in <module>()
      1 visualizer = GQCNNPredictionVisualizer(visualization_config)
----> 2 visualizer.visualize()

/home/aver2/catkin_ws/src/gqcnn/gqcnn/gqcnn_prediction_visualizer.py in visualize(self)
     75             hand_poses_tensor = np.load(os.path.join(self.data_dir, pose_filename))['arr_0']
     76 
---> 77             pose_tensor = self._read_pose_data(hand_poses_tensor, self.input_data_mode)
     78 
     79             # score with neural network

/home/aver2/catkin_ws/src/gqcnn/gqcnn/gqcnn_prediction_visualizer.py in _read_pose_data(self, pose_arr, input_data_mode)
    269         #if input_data_mode == InputDataMode.PARALLEL_JAW:
    270             #return pose_arr[:,2:3]
--> 271         if input_data_mode == InputDataMode.SUCTION:
    272             return np.c_[pose_arr[:,2], pose_arr[:,4]]
    273         elif input_data_mode == InputDataMode.TF_IMAGE:

AttributeError: class InputDataMode has no attribute 'SUCTION'

Thanks in advance for the consideration and help!!!

visatish, 2018-06-28:
Hi @ankurverma-git,

Thanks for pointing this issue out. This was actually a bug where the script was trying to access invalid input data modes. I have pushed a fix for this, can you pull the master branch and retry?

ankurverma-git, 2018-06-29:
Hi @visatish,

Thanks for the help!!! The visualization works correctly now.

Next, I was trying to do grasp planning. I updated the parameters of my policy.yaml file and ran the below code in a new terminal, resulting in the following error: 'ImportError: cannot import name CrossEntropyAntipodalGraspingPolicy'

**Code**

aver2@ubuntu:~/catkin_ws/src/gqcnn$ source devel/setup.bash
aver2@ubuntu:~/catkin_ws/src/gqcnn$ python examples/policy.py --config_filename /home/aver2/catkin_ws/src/gqcnn/cfg/examples/policy.yaml

**Error**

WARNING:root:Failed to import ros dependencies in rigid_transforms.py
/home/aver2/anaconda2/lib/python2.7/site-packages/h5py/__init__.py:36: FutureWarning: Conversion of the second argument of issubdtype from `float` to `np.floating` is deprecated. In future, it will be treated as `np.float64 == np.dtype(float).type`.
  from ._conv import register_converters as _register_converters
WARNING:root:Unable to import pylibfreenect2. Python-only Kinect driver may not work properly.
WARNING:root:Failed to import ROS in Kinect2_sensor.py. Kinect will not be able to be used in bridged mode
WARNING:root:Unable to import openni2 driver. Python-only Primesense driver may not work properly
WARNING:root:Failed to import ROS in primesense_sensor.py. ROS functionality not available
WARNING:root:primesense_sensor.py not installed as catkin package. ROS functionality not available.
WARNING:root:Failed to import ROS in ensenso_sensor.py. ROS functionality not available
WARNING:root:Failed to import ROS in phoxi_sensor.py. PhoXiSensor functionality unavailable.
WARNING:root:Unable to import weight sensor modules!
Traceback (most recent call last):
  File "examples/policy.py", line 127, in <module>
    from gqcnn import CrossEntropyAntipodalGraspingPolicy, RgbdImageState
ImportError: cannot import name CrossEntropyAntipodalGraspingPolicy

I read the earlier issue (no.11) and **tried/know** some things as follows:

1. Catkin_make works correctly and gqcnn gets imported as follows:

aver2@ubuntu:~/catkin_ws$ catkin_make
Base path: /home/aver2/catkin_ws
Source space: /home/aver2/catkin_ws/src
Build space: /home/aver2/catkin_ws/build
Devel space: /home/aver2/catkin_ws/devel
Install space: /home/aver2/catkin_ws/install
####
#### Running command: "make cmake_check_build_system" in "/home/aver2/catkin_ws/build"
####
####
#### Running command: "make -j4 -l4" in "/home/aver2/catkin_ws/build"
####
[  0%] Built target std_msgs_generate_messages_cpp
[  0%] Built target std_msgs_generate_messages_lisp
[  0%] [  0%] Built target sensor_msgs_generate_messages_py
Built target std_msgs_generate_messages_py
[  0%] Built target geometry_msgs_generate_messages_py
[  0%] [  0%] Built target _autolab_core_generate_messages_check_deps_RigidTransformListener
Built target _autolab_core_generate_messages_check_deps_RigidTransformPublisher
[  0%] Built target geometry_msgs_generate_messages_lisp
[  0%] Built target _gqcnn_generate_messages_check_deps_GQCNNGrasp
[  0%] Built target sensor_msgs_generate_messages_lisp
[  0%] [  0%] Built target sensor_msgs_generate_messages_cpp
Built target _gqcnn_generate_messages_check_deps_GQCNNGraspPlanner
[  0%] Built target geometry_msgs_generate_messages_cpp
[  0%] [ 11%] Built target _gqcnn_generate_messages_check_deps_BoundingBox
Built target autolab_core_generate_messages_lisp
[ 22%] Built target autolab_core_generate_messages_cpp
[ 38%] Built target autolab_core_generate_messages_py
[ 55%] [ 83%] Built target gqcnn_generate_messages_lisp
Built target gqcnn_generate_messages_py
[100%] Built target gqcnn_generate_messages_cpp
[100%] Built target autolab_core_generate_messages
[100%] Built target gqcnn_generate_messages
aver2@ubuntu:~/catkin_ws$  source devel/setup.bash
aver2@ubuntu:~/catkin_ws$ python
Python 2.7.15 |Anaconda, Inc.| (default, May  1 2018, 23:32:55) 
[GCC 7.2.0] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import gqcnn
>>>  python examples/policy.py --config_filename /home/aver2/catkin_ws/src/gqcnn/cfg/examples/policy.yaml

2. I have ROS indigo installed.

Thanks in advance for the consideration and help!!!

jeffmahler, 2018-09-18:
Hello, sorry for the delay on this. @ankurverma-git were you able to resolve this issue? I can't see anything that is obviously wrong from the stack trace.

ankurverma-git, 2018-09-19:
Hello Sir,

I could not solve this issue; my work got stuck at that point so I started working on robot control and image processing/transformations, waiting for some assistance. 

Could the error be because of different ROS installed?

Thanks in advance.

jeffmahler, 2018-09-19:
It's possible that it's related to the different version but let's try a few other things:

1. Are you able to import anything from gqcnn? As in, is it just the import of CrossEntropyAntipodalGraspingPolicy that is broken?
2. If you are able to import gqcnn, can you do `import gqcnn; print(gqcnn.__file__)` to make sure it is importing from the correct location?

It would be helpful if you are able to get a stack trace of the ImportError. This may be possible by trying  to import the file gqcnn.policy directly

ankurverma-git, 2018-09-26:
Sorry for the delay Sir, I just finished my mid-semester exams recently.

As per your guidance, I tried the above steps and the results are as follows:

1. I am not able to import anything from gqcnn, i.e. RgbdImageState and Visualizer are also not importing. I commented line 127 and added a new line to import RgbdImageState in policy.py file, which gave the following error:

**Code**

aver2@ubuntu:~/catkin_ws/src/gqcnn$ python examples/policy.py --config_filename /home/aver2/catkin_ws/src/gqcnn/cfg/examples/policy.yaml

**Error**

WARNING:root:Failed to import ros dependencies in rigid_transforms.py
/home/aver2/anaconda2/lib/python2.7/site-packages/h5py/__init__.py:36: FutureWarning: Conversion of the second argument of issubdtype from `float` to `np.floating` is deprecated. In future, it will be treated as `np.float64 == np.dtype(float).type`.
  from ._conv import register_converters as _register_converters
WARNING:root:Unable to import pylibfreenect2. Python-only Kinect driver may not work properly.
WARNING:root:Failed to import ROS in Kinect2_sensor.py. Kinect will not be able to be used in bridged mode
WARNING:root:Unable to import openni2 driver. Python-only Primesense driver may not work properly
WARNING:root:Failed to import ROS in primesense_sensor.py. ROS functionality not available
WARNING:root:primesense_sensor.py not installed as catkin package. ROS functionality not available.
WARNING:root:Failed to import ROS in ensenso_sensor.py. ROS functionality not available
WARNING:root:Failed to import ROS in phoxi_sensor.py. PhoXiSensor functionality unavailable.
WARNING:root:Unable to import weight sensor modules!
Traceback (most recent call last):
  File "examples/policy.py", line 128, in <module>
    from gqcnn import RgbdImageState
ImportError: cannot import name RgbdImageState

The same error occurs when I try to import Visualizer from gqcnn, and also when I try to import either of them directly using the import command.

2. I am able to import gqcnn. The result of "import gqcnn; print(gqcnn.__file__)" is as follows:

import gqcnn; print(gqcnn.__file__)
/home/aver2/catkin_ws/devel/lib/python2.7/dist-packages/gqcnn/__init__.pyc

Please guide me on the last point that you mentioned previously.

Sincere thanks.

visatish, 2019-04-10:
Closing due to inactivity.

closed 2019-04-10 by visatish