BerkeleyAutomation/gqcnn #25: Wrong indentation line 1156 sgd_optimizer.py

mabelzhang, 2018-06-13:
https://github.com/BerkeleyAutomation/gqcnn/blob/be55e0d1576a8f7c312419bcc13019714d0c9128/gqcnn/sgd_optimizer.py#L1156

Seems like this line should definitely be indented to be within the if-statement, otherwise train_image is undefined. It never errors, because gradient_dropout is by default 0 in the YAML.

It also wouldn't error because train_image is defined for a previous distortion above that line, but that's where the behavior gets unknown. It could be setting to the train_image from a previous iteration, or even from the previous distortion.

mabelzhang, 2018-06-13:
Related... same function, different line

https://github.com/BerkeleyAutomation/gqcnn/blob/be55e0d1576a8f7c312419bcc13019714d0c9128/gqcnn/sgd_optimizer.py#L1220

background_denoising block seems to be missing this line at the end, within the for-loop:
`self.train_data_arr[i,:,:,0] = train_image`

Currently train_image is simply discarded after denoising. It is never put back into self.train_data_arr.

visatish, 2018-06-24:
Hi @mabelzhang,

Thanks for using the gqcnn module and for finding these bugs! These definitely should not have been allowed to occur, apologies for that. I will push a fix to the master branch incorporating these changes.

Please let us know if you find anymore bugs like these or if you run into any difficulties!

closed 2018-06-24 by visatish