BerkeleyAutomation/gqcnn #37: CMake Error with catkin_make: version in setup.py (0.0.2) differs from version in package.xml (0.1.0)

salfityj, 2018-09-19:
Trying the GQ-CNN [ROS installation](https://berkeleyautomation.github.io/gqcnn/install/install.html#ros-installation). Coming across an error in mismatch version number upon `catkin_make`, just after `git clone`.

`CMake Error at /opt/ros/kinetic/share/catkin/cmake/catkin_python_setup.cmake: 79: catkin_python_setup() version in setup.py (0.0.2) differs from version in package.xml (0.1.0)`

Currently running ROS Kinetic on Ubuntu 16.04.1

I'll try to play with some version re-numbering in the mean time.

jeffmahler, 2018-09-19:
Sorry about that. Should be fixed with the latest commit!

ZHOUXINWEN, 2018-10-29:
me too ,after I run catkin_make.
CMake Error at /opt/ros/kinetic/share/catkin/cmake/catkin_python_setup.cmake:79 (message):
  catkin_python_setup() version in setup.py (0.0.7) differs from version in
  package.xml (0.0.6)

Currently running ROS Kinetic on Ubuntu 16.04.2

ZHOUXINWEN, 2018-10-31:
I have solved this by comment the 20 line in CMakeLists.txt in autolab_core. the following displayed in the command window

Base path: /home/zxw/catkin_ws
Source space: /home/zxw/catkin_ws/src
Build space: /home/zxw/catkin_ws/build
Devel space: /home/zxw/catkin_ws/devel
Install space: /home/zxw/catkin_ws/install
####
#### Running command: "make cmake_check_build_system" in "/home/zxw/catkin_ws/build"
####
-- Using CATKIN_DEVEL_PREFIX: /home/zxw/catkin_ws/devel
-- Using CMAKE_PREFIX_PATH: /opt/ros/kinetic
-- This workspace overlays: /opt/ros/kinetic
-- Using PYTHON_EXECUTABLE: /usr/bin/python
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/zxw/catkin_ws/build/test_results
-- Found gmock sources under '/usr/src/gmock': gmock will be built
-- Found gtest sources under '/usr/src/gmock': gtests will be built
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.7.14
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 1 packages in topological order:
-- ~~  - autolab_core
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin package: 'autolab_core'
-- ==> add_subdirectory(autolab_core)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- autolab_core: 0 messages, 2 services
-- Configuring done
-- Generating done
-- Build files have been written to: /home/zxw/catkin_ws/build
####
#### Running command: "make -j8 -l8" in "/home/zxw/catkin_ws/build"
####
Scanning dependencies of target std_msgs_generate_messages_lisp
Scanning dependencies of target std_msgs_generate_messages_eus
Scanning dependencies of target _autolab_core_generate_messages_check_deps_RigidTransformPublisher
Scanning dependencies of target _autolab_core_generate_messages_check_deps_RigidTransformListener
Scanning dependencies of target std_msgs_generate_messages_nodejs
Scanning dependencies of target std_msgs_generate_messages_cpp
Scanning dependencies of target std_msgs_generate_messages_py
[  0%] Built target std_msgs_generate_messages_lisp
[  0%] Built target std_msgs_generate_messages_nodejs
[  0%] Built target std_msgs_generate_messages_eus
[  0%] Built target std_msgs_generate_messages_py
[  0%] Built target std_msgs_generate_messages_cpp
[  0%] Built target _autolab_core_generate_messages_check_deps_RigidTransformPublisher
[  0%] Built target _autolab_core_generate_messages_check_deps_RigidTransformListener
Scanning dependencies of target autolab_core_generate_messages_py
Scanning dependencies of target autolab_core_generate_messages_lisp
Scanning dependencies of target autolab_core_generate_messages_nodejs
Scanning dependencies of target autolab_core_generate_messages_eus
Scanning dependencies of target autolab_core_generate_messages_cpp
[ 16%] Generating Python code from SRV autolab_core/RigidTransformPublisher
[ 16%] Generating EusLisp code from autolab_core/RigidTransformPublisher.srv
[ 25%] Generating Lisp code from autolab_core/RigidTransformPublisher.srv
[ 33%] Generating C++ code from autolab_core/RigidTransformPublisher.srv
[ 41%] Generating EusLisp manifest code for autolab_core
[ 50%] Generating Python code from SRV autolab_core/RigidTransformListener
[ 58%] Generating Javascript code from autolab_core/RigidTransformPublisher.srv
[ 66%] Generating EusLisp code from autolab_core/RigidTransformListener.srv
[ 75%] Generating Lisp code from autolab_core/RigidTransformListener.srv
[ 83%] Generating Javascript code from autolab_core/RigidTransformListener.srv
[ 91%] Generating C++ code from autolab_core/RigidTransformListener.srv
[100%] Generating Python srv __init__.py for autolab_core
[100%] Built target autolab_core_generate_messages_nodejs
[100%] Built target autolab_core_generate_messages_lisp
[100%] Built target autolab_core_generate_messages_py
[100%] Built target autolab_core_generate_messages_cpp
[100%] Built target autolab_core_generate_messages_eus

yxmingming, 2019-02-19:
me too 
CMake Error at /opt/ros/kinetic/share/catkin/cmake/catkin_python_setup.cmake:79 (message):
  catkin_python_setup() version in setup.py (0.0.5) differs from version in
  package.xml (0.0.4)

yxmingming, 2019-02-19:
And i also have another question:
After Running Python Scripts, the result is 
Traceback (most recent call last):
  File "examples/policy.py", line 173, in <module>
    sensor = RgbdSensorFactory.sensor(sensor_type, config['sensor'])
  File "/home/yxm/catkin_ws/src/perception/perception/rgbd_sensors.py", line 62, in sensor
    raise ValueError('RGBD sensor type %s not supported' %(sensor_type))
ValueError: RGBD sensor type virtual_primesense not supported

it is right what should i do?
Thanks!

EvanLavizadeh, 2019-03-11:
Same :( Help us Jeff! What is the recommended fix?

jeffmahler, 2019-03-11:
Hey sorry for the delay. That seems to be a broken dependency because gqcnn is version 0.1.0. My best guess is that you have installed BerkeleyAutomation/visualization as a ROS package, because I found that it was broken. Try pulling the latest from origin master in that repo and running catkin_make

visatish, 2019-04-10:
Closing due to inactivity.

closed 2019-04-10 by visatish