BerkeleyAutomation/gqcnn #48: Training Dataset

Bingobingo123, 2019-02-13:
Hello Community,
I would like to train the network myself.

I've seen that on the website there is a repository for the npz file dataset. However there is also an hdf5 file which is also a dataset. Besides, using the dex-net API I can only get hdf5 dataset. What's the difference of these two datasets and how can I train the network with the hdf5 dataset.

Thank you in advance!

visatish, 2019-04-10:
Hi @Bingobingo123,

Sorry for the delayed response. The hdf5 dataset that you can build with the publicly available Dex-Net CLI contains the Dex-Net 1.0 supervisor grasps. These are used as robust supervisor examples during the GQ-CNN training dataset generation, which samples grasps from a joint on-policy distribution of grasps sampled from the policy action space and guiding Dex-Net 1.0 supervisor grasps. You can read more about this distribution in our [latest Dex-Net publication](https://berkeleyautomation.github.io/fcgqcnn/) on on-policy synthetic dataset generation. The resulting dataset is the one with the *.npz files.

Unfortunately, the on-policy dataset generation script is currently not released to the public. If you are interested in generating a dataset for you specific use case, please email jmahler@berkeley.edu.

Thanks,
Vishal

visatish, 2019-04-26:
Closing due to inactivity.

closed 2019-04-26 by visatish