BerkeleyAutomation/gqcnn #50: Grasp sampler does not work with Realsense D435

WrongIndentation, 2019-03-04:
Hi!

I recently tried to use the Realsense D435 depth sensor to sample some grasps canditates. Previously I used the Asus Xtion Pro Live, which was working just fine as I was able to sample grasps with a high grasp quality. For reference: 

Asus Primesense result:
[image]

Realsense result:
[image]

Here's how I got the Images:
```
        ids = discover_cams()
        cfg = {}
        cfg['cam_id'] = ids[0]
        cfg['filter_depth'] = False
        cfg['frame'] = 'realsense_overhead'

        self.sensor = RgbdSensorFactory.sensor('realsense', cfg)
        self.sensor.start()
        self.camera_intr = self.sensor.color_intrinsics

       ..........

        color_im, depth_im, _ = self.sensor.frames()
        frame = self.camera_intr.frame
        color_im = color_im.inpaint(0.5)
        depth_im = depth_im.inpaint(0.5)

        # optionally set a bounding box
        bounding_box = BoundingBox()
        bounding_box.minY = 0
        bounding_box.minX = 0
        bounding_box.maxY = color_im.height
        bounding_box.maxX = color_im.width

        segmask = BinaryImage(255 * np.ones(depth_im.shape).astype(np.uint8),
                              frame=frame)
        try:
            start_time = time.time()
            planned_grasp_data = self.plan_grasp_client(color_im.rosmsg,  # T_grasp_camera
                                                        depth_im.rosmsg,
                                                        self.camera_intr.rosmsg,
                                                        bounding_box,
                                                        segmask.rosmsg)
          ...........
```
The images were taken at roughly the same angle and the same height (about 60 cm).
It seems that the depth is correct, since the pose has a z value of around 60 cm.
What I tried:

- inpaint(1) and no inpaint -> no difference
- different height -> seems to be random
- playing with the depth_scale -> makes it even worse (should be correct)
- using the depth intrinsics `strm = self._profile.get_stream(rs.stream.depth)`  (Line 125 in realsense_sensor.py)  --> scaling seems to be wrong now

I read the other issues but I don't think that I have a scaling issue that's why I am opening a new one.

Hopefully I can get some help here, the depth image looks weird but I don't know why.

Thank you!

jeffmahler, 2019-03-11:
Hey, sorry for this issue. Can you share the images? My best guess is that the depth image is corrupt based on the examples. I don't see the wrench at all, which is fishy.

WrongIndentation, 2019-03-14:
Hey Jeff! Thanks for the feedback :)

I don't actually know what images you excatly want, but I have some .npy files here. 

[Data.zip](https://github.com/BerkeleyAutomation/gqcnn/files/2968845/Data.zip)

This package contains 4 npy files, i tried to use the raw depth values (no filters applied) and scaled/not scaled to meters. 

Asus_Scaled:
[image]
Realsense_Scaled
[image]

The Asus has way clearer edges of the object. Do you think that the realsense is just bad at displaying the edges because it's a stereo camera? Maybe using the function `def median_depth_img(self, num_img=1, fill_depth=0.0)` might help (not tried yet).  I also tried using different presets for the realsense (accuracy, medium density etc.. )

jiuweifox, 2019-03-18:
Maybe you should wait more time to capture data from realSenseD435. The depth image looks like unstable.

visatish, 2019-04-10:
@jeffmahler

WrongIndentation, 2019-04-10:
Hey! Sorry that I haven't made an update. I tried almost everything but in the end I have to say that the Relasense D435 is probably not good enough for this task. 

I skipped frames, took the means of a few depth images, set the visual preset to high accuracy or high/medium density, with/without filter, I even used the ros realsense topics instead of the pyrealsense library but nothing worked. It works sometimes with rather large objects with clear edges. But  small objects ( < ~3 cm) won't work (it does work with the Asus Xtion). 

So if you don't have any other suggestion ( which I will gladly try) you can close this issue since I am out of ideas.

visatish, 2019-04-26:
Closing due to inactivity.

closed 2019-04-26 by visatish

WHGang, 2019-09-08:
Hi, I was wondering how to work with a real camera. Where to set the accept camera topic? Thanks!

vladi315, 2022-06-15:
I had a similar issue with the intel Realsense 435i camera. Grasp sampling worked fine, but the grasp quality was always predicted close to 0. The issue was solved for me after increasing the "crop_height" and "crop_width" parameters in the config file. Now the predicted grasp quality with the realsense reaches up to 0.55.

The reason for this: The resolution of the realsense camera is higher than the resolution of the camera that was used to train the GQ-CNN. Therefore, the cropped 96x96 pixel image that was fed into the network was too small to capture all meaningful features of the object.