BerkeleyAutomation/gqcnn #52: Ros Service Launch Error

jiuweifox, 2019-03-18:
I want to start the serive of gqcnn but failed. The info is like next, 

NODES
  /
    GQCNN_grasp_planner (gqcnn/grasp_planner_node.py)

auto-starting new master
process[master]: started with pid [17722]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 355ddcf4-4949-11e9-9d57-f4d108abcd6d
process[rosout-1]: started with pid [17735]
started core service [/rosout]
Unable to launch [GQCNN_grasp_planner-2]. 
If it is a script, you may be missing a '#!' declaration at the top.
The traceback for the exception was written to the log file
[GQCNN_grasp_planner-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit

Anybody can help me?  Thanks very much

jeffmahler, 2019-03-18:
Try putting `#!/usr/bin/python` at the top of `grasp_planner_node.py`. I'm not sure why it's necessary on your machine though.

jiuweifox, 2019-03-19:
> Try putting `#!/usr/bin/python` at the top of `grasp_planner_node.py`. I'm not sure why it's necessary on your machine though.

Thanks very much!  It worked.

closed 2019-03-19 by jiuweifox