BerkeleyAutomation/gqcnn #55: Error running code : examples/policy.py

Vaibhavjolly, 2019-04-09:
Hi,
I have installed all the required dependencies for running gqcnn code following your GQCNN 1.0.0 documentation .I am trying to run examples/policy.py with dexnet 2.0 pretrained model but getting error  :
1)When I am running the script using python 2.7 : I get different warnings and error as :
./scripts/policies/run_all_dex-net_2.0_examples.sh

RUNNING EXAMPLE 1
WARNING:root:Failed to import geometry msgs in rigid_transformations.py.
WARNING:root:Failed to import ros dependencies in rigid_transforms.py
WARNING:root:autolab_core not installed as catkin package, RigidTransform ros methods will be unavailable
WARNING:root:autolab_perception is not installed as a catkin package - ROS msg conversions will not be available for image wrappers
WARNING:root:autolab_perception is not installed as a catkin package - ROS msg conversions will not be available for image wrappers
WARNING:root:Unable to import CNN modules! Likely due to missing tensorflow.
WARNING:root:TensorFlow can be installed following the instructions in https://www.tensorflow.org/get_started/os_setup
WARNING:root:Unable to import pylibfreenect2. Python-only Kinect driver may not work properly.
WARNING:root:Unable to import openni2 driver. Python-only Primesense driver may not work properly
WARNING:root:Failed to import ROS in primesense_sensor.py. ROS functionality not available
WARNING:root:primesense_sensor.py not installed as catkin package. ROS functionality not available.
WARNING:root:Failed to import ROS in ensenso_sensor.py. ROS functionality not available
Segmentation fault (core dumped)

2)When I am running the script using python 3.6 : I get the following error :
Traceback (most recent call last):
  File "examples/policy.py", line 41, in <module>
    from gqcnn.grasping import RobustGraspingPolicy, CrossEntropyRobustGraspingPolicy, RgbdImageState, FullyConvolutionalGraspingPolicyParallelJaw, FullyConvolutionalGraspingPolicySuction
  File "/home/dl/virtual_env3/lib/python3.7/site-packages/gqcnn-1.0.0-py3.7.egg/gqcnn/__init__.py", line 22, in <module>
ModuleNotFoundError: No module named 'model'

visatish, 2019-04-09:
Hi @Vaibhavjolly,

For the Python 2.7 installation, could you try importing the module in an interpreter and post the output?

Regarding the Python 3.6 installation, we currently only support Python 2.7, thus there could be some package setup differences introduced in Python 3.x that are preventing Python from finding the appropriate modules.

Thanks,
Vishal

Vaibhavjolly, 2019-04-09:
Hi vishal,
Thanks for the quick response ! Yeah,Its not working with python 3.6 but importing with python 2.7 .
Here is the output :  
>>> import model
WARNING:root:Failed to import geometry msgs in rigid_transformations.py.
WARNING:root:Failed to import ros dependencies in rigid_transforms.py
WARNING:root:autolab_core not installed as catkin package, RigidTransform ros methods will be unavailable
root       WARNING  autolab_perception is not installed as a catkin package - ROS msg conversions will not be available for image wrappers
root       WARNING  autolab_perception is not installed as a catkin package - ROS msg conversions will not be available for image wrappers
root       WARNING  Unable to import pylibfreenect2. Python-only Kinect driver may not work properly.
root       WARNING  Unable to import openni2 driver. Python-only Primesense driver may not work properly
root       WARNING  Failed to import ROS in primesense_sensor.py. ROS functionality not available
root       WARNING  primesense_sensor.py not installed as catkin package. ROS functionality not available.
root       WARNING  Failed to import ROS in ensenso_sensor.py. ROS functionality not available
trimesh    WARNING  No FCL -- collision checking will not work
OpenGL.acceleratesupport INFO     OpenGL_accelerate module loaded
OpenGL.arrays.arraydatatype INFO     Using accelerated ArrayDatatype
>>> 
Thanks !

visatish, 2019-04-09:
Oh sorry, I actually meant `import gqcnn`. Could you also provide some system information?

Vaibhavjolly, 2019-04-09:
Hi,
Yeah ! Importing gqcnn gives the following output :
import gqcnn
WARNING:root:Failed to import geometry msgs in rigid_transformations.py.
WARNING:root:Failed to import ros dependencies in rigid_transforms.py
WARNING:root:autolab_core not installed as catkin package, RigidTransform ros methods will be unavailable
root       WARNING  autolab_perception is not installed as a catkin package - ROS msg conversions will not be available for image wrappers
root       WARNING  autolab_perception is not installed as a catkin package - ROS msg conversions will not be available for image wrappers
root       WARNING  Unable to import pylibfreenect2. Python-only Kinect driver may not work properly.
root       WARNING  Unable to import openni2 driver. Python-only Primesense driver may not work properly
root       WARNING  Failed to import ROS in primesense_sensor.py. ROS functionality not available
root       WARNING  primesense_sensor.py not installed as catkin package. ROS functionality not available.
root       WARNING  Failed to import ROS in ensenso_sensor.py. ROS functionality not available
trimesh    WARNING  No FCL -- collision checking will not work
OpenGL.acceleratesupport INFO     OpenGL_accelerate module loaded
OpenGL.arrays.arraydatatype INFO     Using accelerated ArrayDatatype
>>> 
Thanks ! 
Vaibhav

visatish, 2019-04-09:
Hmm, so it seems that everything was installed correctly. Unfortunately, the `Segmentation fault (core dumped)` error message is not very helpful. Have you tried running any of the other example scripts?Can you also try running the training script [here](https://berkeleyautomation.github.io/gqcnn/tutorials/tutorial.html#training)? You don't have to fully train-just check whether or not it runs. My guess is that this is actually coming from some dependency and not the `gqcnn` module itself.

@jeffmahler do you have any thoughts?

Vaibhavjolly, 2019-04-10:
Hey vishal,
The inference code is up now . I installed everything in a new conda environment with python 2.7 and the inference script is working .
Thanks

visatish, 2019-04-10:
Resolved.

Vaibhavjolly, 2019-04-10:
Hi vishal,
Yeah, the inference code worked with python 2.7 .
Thanks !
Vaibhav

WYCEN, 2019-11-22:
> Hi vishal,
> Yeah, the inference code worked with python 2.7 .
> Thanks !
> Vaibhav

hi ,i have the same problem,,how do you solve it?

visatish, 2019-11-22:
Hi @WYCEN,

Which version of Python are you using? Can you also paste the exact output so we can help you debug?

Thanks,
Vishal

WYCEN, 2019-11-22:
> Hi @WYCEN,
> 
> Which version of Python are you using? Can you also paste the exact output so we can help you debug?
> 
> Thanks,
> Vishal

Hi,vishal
I use Python 2.7 on ubuntu16.04，and I use the pip installation,when I evaluate the pre-trained Dex-Net 4.0 PJ network,some errors happened.
[image]

WYCEN, 2019-11-22:
> Hi @WYCEN,
> 
> Which version of Python are you using? Can you also paste the exact output so we can help you debug?
> 
> Thanks,
> Vishal

The steps I have taken are as follows:

$ mkdir myproject
$ cd myproject/
$ virtualenv --no-site-packages venv
$ source venv/bin/activate

$ git clone https://github.com/BerkeleyAutomation/gqcnn.git
$ pip install .

$ ./scripts/downloads/models/download_models.sh
$ python examples/policy.py GQCNN-4.0-PJ --depth_image data/examples/clutter/phoxi/dex-net_4.0/depth_0.npy --segmask data/examples/clutter/phoxi/dex-net_4.0/segmask_0.png --camera_intr data/calib/phoxi/phoxi.intr

visatish, 2019-11-22:
Ahh okay, so it seems that Python 2.7 compatibility is currently broken because of an issue with the `pyglet` dep. It also seems like Python 3.5 compatibility is broken due to a versioning issue with `skimage`. The latest build can be found [here](https://travis-ci.org/BerkeleyAutomation/gqcnn/). However, Python 3.6/3.7 are stable. Can you instead use one of those? I will look into the others, but I don't think I have the bandwidth to do so at the moment. Sorry for the inconvenience!

Thanks,
Vishal

rickstaa, 2019-11-22:
@visatish @WYCEN In my repository for python 2.7 the problems with pyglet and skimage are solved by using scikit-image (0.14.5) and pyglet (1.4.7).

visatish, 2019-11-22:
@rickstaa thanks for the correct pyglet version! The issue I was seeing with scikit-image was that although the GQ-CNN dep explicitly states `scikit-image<=0.14.2`, `pip` does not seem to honor it. Do you know why that might be happening/were you able to find a fix for that? I'm guessing you most likely just manually installed the right version.

WYCEN, 2019-11-22:
> @visatish @WYCEN In my fork the problems with pyglet and skimage are solved for python 2.7 by using scikit-image (0.14.5) and pyglet (1.4.7).
@visatish @rickstaa 
thank you so much,I'll try again.

rickstaa, 2019-11-22:
> image

@visatish I noticed the same behaviour last month so I assumed it had something to do with a hard version dependency of another requirement. In that case, I think at least a requirement conflict warning should have been thrown by pip. I, however, can not remember if such a warning was present. Currently, I do not use your `setup.py` directly. I wrote a parent `setup.py` which installs all requirements of any GitHub submodule that is present.

I did some testing in a `virtualenv` to see what is going on with the gqcnn `setup.py` I however strangely am not able to reproduce the `scikit-image` versioning problem. In a clean python2.7 environment only the pyglet (2.0.dev0) versioning problem is present. I did, however, get the following error when a higher version of `scikit-image` higher than the version you specified is already present on my system:

```
ERROR: gqcnn 1.1.0 has requirement scikit-image<=0.14.2, but you'll have scikit-image 0.14.5 which is incompatible.
```

This is as expected and I think (but am not sure) that by design pip lets it to the user to resolve these kinds of conflicts instead of downgrade or update to the required version itself.

visatish, 2019-11-22:
@rickstaa That's interesting, I might dig into it a bit myself to see if I can replicate the issue then. Thanks so much for debugging for us haha!

rickstaa, 2019-11-22:
@visatish No problem at all. Let me know if you found the issue I'm curious to see what could cause such a problem.

baihaisheng, 2019-11-25:
> @visatish No problem at all. Let me know if you found the issue I'm curious to see what could cause such a problem.
@ rickstaa 
Hi,I have installed pyglet (1.4.7) under python 2.7,but some other errors have arisen as follow：
[image]
thanks

visatish, 2019-11-25:
Hi @baihaisheng,

This is a fairly common TensorFlow error that can be caused by a few different things under-the-hood. Off the top of my head, it could be a potential version issue between TensorFlow and CUDA. I'm afraid I won't be able to help too much with it, but you can definitely find similar errors if you Google a bit!

Thanks,
Vishal

baihaisheng, 2019-11-25:
Thank you so much,I'll check the problem and solve it.

baihaisheng, 2019-11-25:
> Hi @baihaisheng,
>
> This is a fairly common TensorFlow error that can be caused by a few different things under-the-hood. Off the top of my head, it could be a potential version issue between [...]

 Hi @visatish 
I install the tensorflow-gpu 1.7.0 and the problem has been solved.
when I  run pip install .，The system is installed by default tensorflow-gpu 1.13.1,cause a +version issue between TensorFlow and CUDA.

shineYuSong, 2022-11-22:
> Resolved.

i have the same question, and can you share the solution? thanks in advance !

baihaisheng, 2022-11-22:
你好，你的邮件已经收到！祝好！柏海生