BerkeleyAutomation/gqcnn #60: Issue: Other Issue

futurev, 2019-04-18:
**System information**
- OS Platform and Distribution (e.g., Linux Ubuntu 16.04): windows10
- Python version: python3.6
- Installed using pip or ROS: pip
- GPU model (if applicable):

**Describe the result you are trying to replicate**
If you can, provide a link to the section in the [documentation](https://berkeleyautomation.github.io/gqcnn/index.html).

**Provide the exact sequence of commands / steps that you executed to replicate this result**

**Describe the unexpected behavior** 

**Other info / logs**
Include any logs or source code that would be helpful to diagnose the problem. If including tracebacks, please include the full traceback. Large logs and files should be attached.

In the paper, Fig2 shows that fc3's kernel shapes is [18, 18], but when I run the code, I got filter_dim for fc3 [17, 17] in fc_network_tf.py--> _build_im_stream(), from input size 96*96--kernel (9, 9)--kernel (5, 5) --kernel (5, 5) --kernel (5, 5) 
Could you please confirm which one is correct?

visatish, 2019-04-18:
Hi @futurev,

What you are seeing at runtime is correct. There was an error in the paper calculation where we didn't use floor div, hence the off-by-one.

Thanks,
Vishal

futurev, 2019-04-22:
same thinking, thanks

closed 2019-04-26 by visatish