BerkeleyAutomation/gqcnn #61: Issue: Installation Issue

ChanWaiYang, 2019-04-22:
**System information**
- OS Platform and Distribution (e.g., Linux Ubuntu 16.04): Ubuntu 16.04
- Python version: python 2.7
- Installed using pip or ROS: ROS installation

Hi,

I have finish installed the installation for the GQCNN for ROS installation and tested the python policy with pre-trained network and sample image using the following command included in documentation with no issue
> $ python examples/policy.py GQCNN-4.0-PJ --depth_image data/examples/clutter/phoxi/dex-net_4.0/depth_0.npy --segmask data/examples/clutter/phoxi/dex-net_4.0/segmask_0.png --camera_intr data/calib/phoxi/phoxi.intr

However, I had issue with testing the ROS policy and the command that I used are:
> $ roslaunch gqcnn grasp_planning_service.launch model_name:=GQCNN-4.0-PJ

> $ python examples/policy_ros.py --depth_image data/examples/clutter/phoxi/dex-net_4.0/depth_0.npy --segmask data/examples/clutter/phoxi/dex-net_4.0/segmask_0.png --camera_intr data/calib/phoxi/phoxi.intr

I modified the config_filename in grasp_planner_node.py line 333 to read the ros_node yaml from 'cfg/policies/gqcnn_pj.yaml' to 'cfg/ros_nodes/grasp_planner_node.yaml'

The error i encountered with:

> [INFO] [1555922099.759817]: Planning Grasp
[ERROR] [1555922099.800620]: Error processing request: 'rgbd_state'
['Traceback (most recent call last):\n', '  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 625, in _handle_request\n    response = convert_return_to_response(self.handler(request), self.response_class)\n', '  File "/home/junri/catkin_gqcnn/src/gqcnn/ros_nodes/grasp_planner_node.py", line 122, in plan_grasp\n    return self._plan_grasp(color_im, depth_im, camera_intr)\n', '  File "/home/junri/catkin_gqcnn/src/gqcnn/ros_nodes/grasp_planner_node.py", line 221, in _plan_grasp\n    if self.cfg[\'vis\'][\'rgbd_state\']:\n', "KeyError: 'rgbd_state'\n"]
^C[gqcnn/grasp_planner-2] killing on exit

From the error, it seems like the processing is looking for vis - rgbd_state parameter from yaml but this vis-rgbd_state parameter is missing in the cfg/ros_nodes/grasp_planner_node.yaml

Am I suppose to keep on adding the missing parameters to the yaml?

visatish, 2019-04-22:
Hi @ChanWaiYang,

Sorry about that, there were some things broken in our ROS policy that have now been patched. The patch is waiting approval to be merged into `master`, but you can check out `ros_policy_patch` for now. You will also have to upgrade the `perception` package as a new version was pushed to PyPi with some necessary updates. 

Please let me know if you are able to run things now!

Thanks,
Vishal

jeffmahler, 2019-04-22:
@visatish @ChanWaiYang I just approved and merged the PR. It looks like both the config filenames and the image encodings have been properly fixed so hopefully things will work now.

ChanWaiYang, 2019-04-24:
@visatish @jeffmahler Thank for the fix. Issue solved.

visatish, 2019-04-24:
Awesome!

closed 2019-04-24 by visatish