BerkeleyAutomation/gqcnn #68: Issue: Bug/Performance Issue [Replication]

futurev, 2019-04-30:
**System information**
- OS Platform and Distribution (e.g., Linux Ubuntu 16.04): windows10
- Python version: python3.6
- Installed using pip or ROS: no ros
- GPU model (if applicable): 

**Describe the result you are trying to replicate**
If you can, provide a link to the section in the [documentation](https://berkeleyautomation.github.io/gqcnn/index.html).

**Provide the exact sequence of commands / steps that you executed to replicate this result**
gqcnn/tools/finetune.py

**Describe the unexpected behavior** 
Traceback (most recent call last):
GQCNNTrainerTF INFO     Cleaning and preparing to exit optimization...
  File "C:/cProjects/picking/fcnn/gqcnn/tools/finetune.py", line 140, in <module>
GQCNNTrainerTF INFO     Terminating prefetch queue workers...
    trainer.finetune(model_dir)
GQCNNTrainerTF INFO     Flushing prefetch queue...
  File "C:\cProjects\picking\fcnn\gqcnn\gqcnn\training\tf\trainer_tf.py", line 217, in finetune
GQCNNTF    INFO     Closing TF Session...
    self._finetune(base_model_dir)
  File "C:\cProjects\picking\fcnn\gqcnn\gqcnn\training\tf\trainer_tf.py", line 241, in _finetune
    self._optimize_weights(finetune=True)
  File "C:\cProjects\picking\fcnn\gqcnn\gqcnn\training\tf\trainer_tf.py", line 320, in _optimize_weights
    p.start()
  File "C:\Users\ecl\AppData\Local\Continuum\miniconda3\envs\py36tf\lib\multiprocessing\process.py", line 105, in start
    self._popen = self._Popen(self)
  File "C:\Users\ecl\AppData\Local\Continuum\miniconda3\envs\py36tf\lib\multiprocessing\context.py", line 223, in _Popen
    return _default_context.get_context().Process._Popen(process_obj)
  File "C:\Users\ecl\AppData\Local\Continuum\miniconda3\envs\py36tf\lib\multiprocessing\context.py", line 322, in _Popen
    return Popen(process_obj)
  File "C:\Users\ecl\AppData\Local\Continuum\miniconda3\envs\py36tf\lib\multiprocessing\popen_spawn_win32.py", line 65, in __init__
    reduction.dump(process_obj, to_child)
  File "C:\Users\ecl\AppData\Local\Continuum\miniconda3\envs\py36tf\lib\multiprocessing\reduction.py", line 60, in dump
    ForkingPickler(file, protocol).dump(obj)
TypeError: can't pickle _thread.RLock objects
WARNING:root:Failed to import geometry msgs in rigid_transformations.py.
WARNING:root:Failed to import ros dependencies in rigid_transforms.py
WARNING:root:autolab_core not installed as catkin package, RigidTransform ros methods will be unavailable

Process finished with exit code 1

**Other info / logs**
Include any logs or source code that would be helpful to diagnose the problem. If including tracebacks, please include the full traceback. Large logs and files should be attached.

visatish, 2019-04-30:
Hi @futurev,

We *do not support* python3.6, and there are some known incompatibilities. Please try again with python2.7 and let us know if the issue persists. 

Thanks,
Vishal

futurev, 2019-04-30:
Thank you for the quick response. but do you have any clue about why this happened in Python3.6?

visatish, 2019-04-30:
It seems to be an error inside the `multiprocessing` library when trying to serialize the thread lock. I'm not entirely sure why this would happen in the first place/only error out in Python 3.6 though. It's possible that something else is wrong with your installation. We also have not tested the library on Windows, only on Ubuntu, so there could be something there.

visatish, 2019-05-31:
Hi @futurev,

Were you able to fix this? We just pushed **v1.1.0**, which is compatible with Python 3.5/3.6/3.7. Maybe you can try that, although it has still only been tested on Ubuntu.

Thanks,
Vishal

visatish, 2019-06-12:
Closing due to inactivity.

closed 2019-06-12 by visatish