BerkeleyAutomation/gqcnn #69: Issue: Installation Issue with pip

nekovfra, 2019-05-20:
**System information**
- OS Platform and Distribution Linux Ubuntu 16.04 in VMWare
- Python version: 2.7.12
- Installed using pip or ROS: pip

**Describe the problem**
pip natively downloads python3 modules, specifying python27 versions in install.py file doesn't help because of nested module dependency.
Running on clean install of OS.

**Provide the exact sequence of commands / steps that you executed before running into the problem**
pip install .

**Any other info / logs**
Include any logs or source code that would be helpful to diagnose the problem. If including tracebacks, please include the full traceback. Large logs and files should be attached.

....
    Running scikit-image-0.15.0/setup.py -q bdist_egg --dist-dir /tmp/easy_install-ZV5iUC/scikit-image-0.15.0/egg-dist-tmp-C93hgK
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-build-5LtMVA/meshrender/setup.py", line 47, in <module>
        'sphinx_rtd_theme'
      File "/usr/lib/python2.7/distutils/core.py", line 111, in setup
        _setup_distribution = dist = klass(attrs)
      File "/usr/lib/python2.7/dist-packages/setuptools/dist.py", line 269, in __init__
        self.fetch_build_eggs(attrs['setup_requires'])
      File "/usr/lib/python2.7/dist-packages/setuptools/dist.py", line 313, in fetch_build_eggs
        replace_conflicting=True,
      File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 826, in resolve
        dist = best[req.key] = env.best_match(req, ws, installer)
      File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 1092, in best_match
        return self.obtain(req, installer)
      File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 1104, in obtain
        return installer(requirement)
      File "/usr/lib/python2.7/dist-packages/setuptools/dist.py", line 380, in fetch_build_egg
        return cmd.easy_install(req)
      File "/usr/lib/python2.7/dist-packages/setuptools/command/easy_install.py", line 663, in easy_install
        return self.install_item(spec, dist.location, tmpdir, deps)
      File "/usr/lib/python2.7/dist-packages/setuptools/command/easy_install.py", line 693, in install_item
        dists = self.install_eggs(spec, download, tmpdir)
      File "/usr/lib/python2.7/dist-packages/setuptools/command/easy_install.py", line 873, in install_eggs
        return self.build_and_install(setup_script, setup_base)
      File "/usr/lib/python2.7/dist-packages/setuptools/command/easy_install.py", line 1101, in build_and_install
        self.run_setup(setup_script, setup_base, args)
      File "/usr/lib/python2.7/dist-packages/setuptools/command/easy_install.py", line 1087, in run_setup
        run_setup(setup_script, args)
      File "/usr/lib/python2.7/dist-packages/setuptools/sandbox.py", line 246, in run_setup
        raise
      File "/usr/lib/python2.7/contextlib.py", line 35, in __exit__
        self.gen.throw(type, value, traceback)
      File "/usr/lib/python2.7/dist-packages/setuptools/sandbox.py", line 195, in setup_context
        yield
      File "/usr/lib/python2.7/contextlib.py", line 35, in __exit__
        self.gen.throw(type, value, traceback)
      File "/usr/lib/python2.7/dist-packages/setuptools/sandbox.py", line 166, in save_modules
        saved_exc.resume()
      File "/usr/lib/python2.7/dist-packages/setuptools/sandbox.py", line 141, in resume
        six.reraise(type, exc, self._tb)
      File "/usr/lib/python2.7/dist-packages/setuptools/sandbox.py", line 154, in save_modules
        yield saved
      File "/usr/lib/python2.7/dist-packages/setuptools/sandbox.py", line 195, in setup_context
        yield
      File "/usr/lib/python2.7/dist-packages/setuptools/sandbox.py", line 243, in run_setup
        DirectorySandbox(setup_dir).run(runner)
      File "/usr/lib/python2.7/dist-packages/setuptools/sandbox.py", line 273, in run
        return func()
      File "/usr/lib/python2.7/dist-packages/setuptools/sandbox.py", line 242, in runner
        _execfile(setup_script, ns)
      File "/usr/lib/python2.7/dist-packages/setuptools/sandbox.py", line 46, in _execfile
        exec(code, globals, locals)
      File "/tmp/easy_install-ZV5iUC/scikit-image-0.15.0/setup.py", line 31, in <module>
        classifiers = [
      File "/home/user/.local/lib/python2.7/site-packages/numpy/__init__.py", line 142, in <module>
        from . import core
      File "/home/user/.local/lib/python2.7/site-packages/numpy/core/__init__.py", line 40, in <module>
        from . import multiarray
      File "/home/user/.local/lib/python2.7/site-packages/numpy/core/multiarray.py", line 12, in <module>
        from . import overrides
      File "/home/user/.local/lib/python2.7/site-packages/numpy/core/overrides.py", line 46, in <module>
        """)
    RuntimeError: implement_array_function method already has a docstring

visatish, 2019-05-23:
Hi @nekovfra,

Sorry you're having trouble with this. We're in the middle of porting the `gqcnn` library over to Python 3, which should solve a lot of these problems as Python 2 is quickly losing support in many of our dependencies. This should be done by early next week, so I would advise you to check back then!

Thanks,
Vishal

nekovfra, 2019-05-23:
Thank you for your swift answer, you're amazing. I'll definitely check the project out later. Good luck with the porting.

visatish, 2019-05-30:
Hi @nekovfra,

Sorry for the delay! We just pushed v1.1.0, which is compatible with Python 3.5/3.6/3.7. We have fixed the Python 2.7 installation, but I would strongly encourage you to try out the Python 3 one. Let me know if you run into any issues, and once again thanks for you interest!

Thanks,
Vishal

rickstaa, 2019-06-05:
I just tested the fix and both the `pip install .` and `pip3 install .` command work on my system.

**System information:**
**OS:**Ubuntu 16.04
**python2 version:** 2.7.12
**python3 version:** 3.5.2
**pip version:** 19.1.1
**pip3 version:** 19.1.1
**python install method:** Included with the OS
**pip install method:** Installed using the following commands (due to the `ImportError: cannot import name main` error while installing using apt):

```
curl https://bootstrap.pypa.io/get-pip.py | python
curl https://bootstrap.pypa.io/get-pip.py | python3
```

visatish, 2019-06-05:
Awesome, great to hear that @rickstaa!

closed 2019-06-06 by nekovfra