BerkeleyAutomation/gqcnn #70: Issue: Installation Issue - Wrong setup.py pip dependencies

rickstaa, 2019-05-22:
**System information**
- OS Platform and Distribution (e.g., Linux Ubuntu 16.04):
- Python version: 2.7.5
- Installed using pip or ROS: pip

**Describe the problem**
When Installing the python package I get a number of dependency errors:

```ERROR: autolab-perception 0.0.7 has requirement matplotlib<=2.2.0, but you'll have matplotlib 2.2.4 which is incompatible.
ERROR: visualization 0.0.6 has requirement matplotlib<=2.2.0, but you'll have matplotlib 2.2.4 which is incompatible.
ERROR: autolab-core 0.0.11 has requirement matplotlib<=2.2.0, but you'll have matplotlib 2.2.4 which is incompatible.
```

I now changed your *setup.py requirements* from:

```
requirements = [
    'autolab-core',
    'autolab-perception',
    'visualization',
    'numpy>=1.14.0',
    'scipy',
    'matplotlib<3.0.0',
    'opencv-python',
    'scikit-image<0.15.0',
    'scikit-learn',
    'psutil',
    'gputil'
]
```

to:

```
requirements = [
    'autolab-core',
    'autolab-perception',
    'visualization',
    'numpy>=1.14.0',
    'scipy',
    'matplotlib<2.2.0',
    'opencv-python',
    'scikit-image<0.15.0',
    'scikit-learn',
    'psutil',
    'gputil'
]
```

This could be due to my specific setup but I thought you should know.

**Provide the exact sequence of commands / steps that you executed before running into the problem**
1. Download the *gqcnn* repository.
2. Inside the *gqcnn* folder run the `pip install .` command

jeffmahler, 2019-05-22:
@rickstaa Thanks for bringing up this issue.

@mmatl @visatish Do you know why meshrender and autolab_core depend on matplotlib 2.2.0? I don't see any reason not to upgrade but there must be some problem.

mmatl, 2019-05-22:
Matplotlib > 2.2.0 breaks in Python3.

visatish, 2019-05-23:
Hi @rickstaa,

Sorry you're having trouble with this. We're in the middle of porting the `gqcnn` library over to Python 3, which should solve a lot of these problems as in doing so we are cleaning up/fixing many installation dependencies. This should be done by early next week, so I would advise you to check back then if you can't get it working in the current state.

Thanks,
Vishal

rickstaa, 2019-05-24:
@visatish Thanks for pointing that out. I will check the repository next week as kept running into conflicting dependencies with the current versions of `gqcnn`, `autolab_perception` and `autolab_core`. Looking forward to your new version :).

visatish, 2019-05-30:
Hi @rickstaa ,

Sorry for the delay! We just pushed **v1.1.0**, which is compatible with Python 3.5/3.6/3.7. We have fixed the Python 2.7 installation, but I would strongly encourage you to try out the Python 3 one. Let me know if you run into any issues, and once again thanks for you interest!

Thanks,
Vishal

rickstaa, 2019-06-04:
Similar to #69 thus closing my issue.

closed 2019-06-04 by rickstaa