BerkeleyAutomation/gqcnn #72: Issue: Bug/Performance Issue [Replication]

visatish, 2019-05-28:
**System information**
- OS Platform and Distribution (e.g., Linux Ubuntu 16.04): Linux Ubuntu 16.01
- Python version: 3.5.2
- Installed using pip or ROS: pip
- GPU model (if applicable): N/A

**Describe the result you are trying to replicate**
If you can, provide a link to the section in the [documentation](https://berkeleyautomation.github.io/gqcnn/index.html).
Test Dex-Net 4.0 w/ Suction.

**Provide the exact sequence of commands / steps that you executed to replicate this result**
`./scripts/policies/run_all_dex-net_4.0_suction_examples.sh`

**Describe the unexpected behavior** 
Errors out.

**Other info / logs**
Include any logs or source code that would be helpful to diagnose the problem. If including tracebacks, please include the full traceback. Large logs and files should be attached.
`Traceback (most recent call last):
  File "examples/policy.py", line 260, in <module>
    action = policy(state)
  File "/home/vsatish/Workspace/gqcnn_release/gqcnn/gqcnn/grasping/policy/policy.py", line 228, in __call__
    return self.action(state)
  File "/home/vsatish/Workspace/gqcnn_release/gqcnn/gqcnn/grasping/policy/policy.py", line 405, in action
    action = self._action(state)
  File "/home/vsatish/Workspace/gqcnn_release/gqcnn/gqcnn/grasping/policy/policy.py", line 1287, in _action
    grasps, q_values = self.action_set(state)
  File "/home/vsatish/Workspace/gqcnn_release/gqcnn/gqcnn/grasping/policy/policy.py", line 1172, in action_set
    axis=grasp_axis)
  File "/home/vsatish/Workspace/gqcnn_release/gqcnn/gqcnn/grasping/grasp.py", line 376, in from_feature_vec
    if v.shape > 2 and axis is None:
TypeError: unorderable types: tuple() > int()`

visatish, 2019-05-28:
@jeffmahler Is this a bug we never caught because in Python 2.7 it always returns `True`?

jeffmahler, 2019-05-29:
@visatish yes, that's probably why it worked before. I'm pretty sure there just needs to be a len() call around the shape. I'll make a quick PR when I'm at my desktop.

visatish, 2019-05-29:
Oh okay, if that's what you think it is, I can patch it. I just looked at it once and was like "I have no idea what this is supposed to do." haha.

jeffmahler, 2019-05-29:
Ok I think this is fixed in #73

visatish, 2019-05-30:
Fixed in https://github.com/BerkeleyAutomation/gqcnn/pull/74.

closed 2019-05-30 by visatish

baihaisheng, 2019-12-09:
> Ok I think this is fixed in #73
Hi, Mr.mahler
I am a student from China and impressed by DEX-NET. I have already finished the tutorial and it works well. Now I want to use the KinectV2 to detect the grasping pose and send a message to UR5 robot by ROS. Unfortunately, I have some problems with the modification of examples/policy_ros.py, can you give me some suggestions? thanks a lot.
best wishes.