BerkeleyAutomation/gqcnn #75: Python3 policy installation and execution issues

rickstaa, 2019-06-18:
# System information
OS: Ubutun 16.04
Container: Yes created with singularity 3.2.1
GQCCN installation: Installed with pip
Python version: 3.7
Virtual environment: Both tested in [conda](https://docs.conda.io/en/latest/) and [Virtualenv](https://virtualenv.pypa.io/en/stable/).

# Problem description
I tried installing and running the GQCNN python3 policy both in a python virtual environment as well as in a conda environment. However, while doing this on a fresh ubuntu 16.04 installation I encounter a number of problems. For documentation, I listed the problems I encountered plus the solution below. 

# GQCNN python policy inside a python virtual environment (virtualenv)

## Pip installation problems

### Meshrender - Installation error
During the pip installation i get a `ERROR: Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-install-nks1vqne/meshrender/
`. The full error log can be found [here](https://gist.github.com/rickstaa/7efb3026fb501ded163e592a13942d36).

#### Solution
This was caused by the python3 version of python-dev not being present on my system this can be solved by using the `apt install python-dev` command.

### Meshrender - pyglet dependency error
The when running the `setup.py` through pip the [pyglet](https://bitbucket.org/pyglet/pyglet/wiki/Home) package unfortunatelly is not updated to the right version. As a result we get the `ERROR: meshrender 0.0.10 has requirement pyglet>=1.4.0b1, but you'll have pyglet 1.3.2 which is incompatible` error.

#### Solution
This can be solved by manually installing pyglet (v1.4.0b1) using the `pip install pyglet==1.4.0b1` command.

## Python policy execution problems
To test the policy I downloaded the trained models using the `./scripts/downloads/models/download_models.sh` command. Following I tested the python policy running the following command:

```
python3 examples/policy.py GQCNN-2.0 --depth_image data/examples/single_object/primesense/depth_0.npy --segmask data/examples/single_object/primesense/segmask_0.png --camera_intr data/calib/phoxi/phoxi.intr
```

### _tkinter module error
```
Traceback (most recent call last):
  File "/usr/lib/python3.5/tkinter/__init__.py", line 36, in <module>
    import _tkinter
ImportError: No module named '_tkinter'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "examples/policy.py", line 46, in <module>
    from perception import (BinaryImage, CameraIntrinsics, ColorImage, DepthImage,
  File "/root/virtualenviroments/gqcnn/lib/python3.5/site-packages/perception/__init__.py", line 8, in <module>
    from .camera_intrinsics import CameraIntrinsics
  File "/root/virtualenviroments/gqcnn/lib/python3.5/site-packages/perception/camera_intrinsics.py", line 14, in <module>
    from .image import DepthImage, PointCloudImage
  File "/root/virtualenviroments/gqcnn/lib/python3.5/site-packages/perception/image.py", line 12, in <module>
    import matplotlib.pyplot as plt
  File "/root/virtualenviroments/gqcnn/lib/python3.5/site-packages/matplotlib/pyplot.py", line 115, in <module>
    _backend_mod, new_figure_manager, draw_if_interactive, _show = pylab_setup()
  File "/root/virtualenviroments/gqcnn/lib/python3.5/site-packages/matplotlib/backends/__init__.py", line 62, in pylab_setup
    [backend_name], 0)
  File "/root/virtualenviroments/gqcnn/lib/python3.5/site-packages/matplotlib/backends/backend_tkagg.py", line 4, in <module>
    from . import tkagg  # Paint image to Tk photo blitter extension.
  File "/root/virtualenviroments/gqcnn/lib/python3.5/site-packages/matplotlib/backends/tkagg.py", line 5, in <module>
    from six.moves import tkinter as Tk
  File "/root/virtualenviroments/gqcnn/lib/python3.5/site-packages/six.py", line 92, in __get__
    result = self._resolve()
  File "/root/virtualenviroments/gqcnn/lib/python3.5/site-packages/six.py", line 115, in _resolve
    return _import_module(self.mod)
  File "/root/virtualenviroments/gqcnn/lib/python3.5/site-packages/six.py", line 82, in _import_module
    __import__(name)
  File "/usr/lib/python3.5/tkinter/__init__.py", line 38, in <module>
    raise ImportError(str(msg) + ', please install the python3-tk package')
ImportError: No module named '_tkinter', please install the python3-tk package
```

#### Solution
Install the `python3-tk` package using the `apt-get install python3-tk` command.

# GQCNN python policy package inside a conda enviroment (conda)

## pip installation problems

### Prompt_toolkit version problem

```
ERROR: jupyter-console 6.0.0 has requirement prompt_toolkit<2.1.0,>=2.0.0, but you'll have prompt-toolkit 1.0.16 which is incompatible.
```

#### Solution
We can install the right version using `pip install prompt_toolkit==2.0.9` but then we get a `ERROR: ipython 5.5.0 has requirement prompt-toolkit<2.0.0,>=1.0.4, but you'll have prompt-toolkit 2.0.9 which is incompatible` error. Therefore I decided to leave `prompt_toolkit` to version 1.0.4.

### Meshrender installation error
During the pip installation i get a `ERROR: Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-install-nks1vqne/meshrender/
`. The full error log can be found [here](https://gist.github.com/rickstaa/7efb3026fb501ded163e592a13942d36).`

## Python policy execution problems
To test the policy I downloaded the trained models using the `./scripts/downloads/models/download_models.sh` command. Following I tested the python policy running the following command:

``` 
python3 examples/policy.py GQCNN-2.0 --depth_image data/examples/single_object/primesense/depth_0.npy --segmask data/examples/single_object/primesense/segmask_0.png --camera_intr data/calib/phoxi/phoxi.intr
```

#### Solution
This was caused by the python3 version of python-dev not being present on my system this can be solved by using the `apt install python-dev` command.

### Solution
Install the right version of prompt_toolkit using `conda install prompt_toolkit`.

# Other possible errors

## pip Installation problems

### Python package dependency problem
First when I try to install the `GQCNN` package using `catkin build` I get a lot of dependency errors. These errors were caused by the fact that I did not run the `setup.py` before building the ROS package. It might be an idea to add this step to the [gqcnn documentation](https://berkeleyautomation.github.io/gqcnn/).

#### Solution
1. Go into the `GQCNN` main folder and execute the following command:

- `python3 -m pip install .`

##### Pip update warning
When I try to install the package I get the following a `you are using pip version 8.1.1, however, version 19.1.1 is available` warning. This can be solved by updating pip using the `pip install --upgrade pip`. On some systems, due to an error introduced in pip 10 ([see this topic](https://stackoverflow.com/questions/28210269/importerror-cannot-import-name-main-when-running-pip-version-command-in-windo)) you need to install pip using the get-pip.py file instead of using the apt repository. This can be done using as explained below:

`curl https://bootstrap.pypa.io/get-pip.py -o get-pip.py && python3 get-pip.py && rm get-pip.py`

visatish, 2019-06-18:
Hi @rickstaa,

Thank you for writing this detailed report with solutions included :) A lot of the issues here seem to be missing system-wide deps. The reason you encountered them with Python3 and not with Python2 is probably because Python2 is the default on Ubuntu 16.04 and thus all the deps for it are there already (`python-`). We install all of these manually in our docker files [here](https://github.com/BerkeleyAutomation/gqcnn/tree/master/docker).

The `pip` issue is also addressed the same way in the docker file, however we have just been ignoring the `pyglet` error since it does not actually break anything down the road (as far as I have seen).

Thanks,
Vishal

rickstaa, 2019-06-19:
@visatish Ah thanks for the response. I will close the issue.

closed 2019-06-19 by rickstaa