BerkeleyAutomation/gqcnn #77: Issue: Bug/Performance Issue [Physical Robot]

rickstaa, 2019-06-21:
# System information
- OS Platform and Distribution (e.g., Linux Ubuntu 16.04): Ubuntu 16.04 singularity v3.2.1 container
- Python version: 2.7.12 & 3.5.2
- Installed using pip or ROS: ROS
- Camera: Kinect v2 (Libfreenect2 library)
- Gripper: Parallel jaw gripper
- Robot: Emika Franka Panda robot
- GPU model (if applicable): Nvidia M1000M GPU

# Describe what you are trying to do
I am trying to test your GQCNN ROS package on our Emika Franka Panda robot.

# Problem description
Before applying it on the real robot I first am testing out your package inside a gazebo simulation. I tried this both in a python 2.7 and python 3.6 enviroment. 

## Python 2.7.15 virtual enviroment (virtualenv package)
When I try to launch the ROS node (After building with `catkin_make or catkin build`) using the `roslaunch gqcnn grasp_planning_service.launch model_name:=GQCNN-4.0-PJ` command I get the following error:

```
Traceback (most recent call last):
  File "/catkin_ws/src/gqcnn/ros_nodes/grasp_planner_node.py", line 45, in <module>
    from autolab_core import YamlConfig
ImportError: No module named autolab_core
```

Although I could not find it in the documentation I assume this is caused by me not having installed the packages specified in the `setup.py` file. I, therefore, created a python 2.7.15 virutal enviroment sourced the enviroment and executed the following commands:

`python -m pip install -r gqcnn/requirements/p2/gpu_requirements.txt`

However if I then want to build the ROS package again I logically get the following error:

```
ImportError: "from catkin_pkg.package import parse_package" failed: No module named catkin_pkg.package
Make sure that you have installed "catkin_pkg", it is up to date and on the PYTHONPATH.
CMake Error at /opt/ros/kinetic/share/catkin/cmake/safe_execute_process.cmake:11 (message):
  execute_process(/virtualenvs/test/bin/python
  "/opt/ros/kinetic/share/catkin/cmake/parse_package_xml.py"
  "/opt/ros/kinetic/share/catkin/cmake/../package.xml"
  "/catkin_ws/build/catkin/catkin_generated/version/package.cmake") returned
  error code 1
Call Stack (most recent call first):
  /opt/ros/kinetic/share/catkin/cmake/catkin_package_xml.cmake:74 (safe_execute_process)
  /opt/ros/kinetic/share/catkin/cmake/all.cmake:167 (_catkin_package_xml)
  /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:20 (include)
  CMakeLists.txt:56 (find_package)
```
This is logical since the rospackages are installed on the global python 2.7.15 interperter. I therefore deactivate the virtual env build the ROS package and resource the virtual env. When I then try to launch the ROS package in the virtual enviroment I not suprisingly get the following error:

```
process[gqcnn/grasp_planner-2]: started with pid [1701]
Traceback (most recent call last):
  File "/catkin_ws/src/gqcnn/ros_nodes/grasp_planner_node.py", line 43, in <module>
    import rospy
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/__init__.py", line 49, in <module>
    from .client import spin, myargv, init_node, \
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/client.py", line 52, in <module>
    import roslib
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslib/__init__.py", line 50, in <module>
    from roslib.launcher import load_manifest
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslib/launcher.py", line 42, in <module>
    import rospkg
ImportError: No module named rospkg
```

This again is caused by the rospackages being installed in the global python interperter. This behavoir is explained on [this topic](https://answers.ros.org/question/244644/conflict-with-ros-python-and-conda/). I tried to solve this by running the following commands:

```
pip install rospkg
``` 
However, I now got a lot of other errors. So I'm unsure if this is the way to go. Can you maybe alloberate on how the ROS package should be used. Does it need to run without installing the python packages, does it need to run inside a python 2.7.15 enviroment since ros is in this enviroment?

visatish, 2019-06-23:
Hi @rickstaa,

Sorry to hear that you're having problems with the ROS setup. As a disclaimer, we do not use ROS internally as there is significant communications overhead, and ROS has turned out to be unnecessarily involved in general haha. Thus, the ROS package was really just created in case others wanted to use it, and thus it is a bit buggy due to a lack of rigorous testing. I can share with you what I normally do to get it running:

*ROS Kinetic has been installed system-wide on the development machine.

1) Create a new virtual env (I would stick to *Python2.7* here because I actually never got around to testing the ROS capabilities on Python3. I will do so when I get a chance.). This virtual environment should have access to system-wide packages. (I believe it's the default behaviour, but I also think that it can be disabled.)

2) Install all packages through the appropriate `requirements.txt`. (You are correct that this needs to be done beforehand since ROS normally uses `rosdep` for dependencies but we didn't want to go with that since some libraries aren't supported in their registry. I will update the installation instructions.)

3) Run `catkin make` and *source `devel/setup.bash`*.

4) You might also have to `pip install` some libraries here and there like `rospkg` and `rospy`. This is just due to a lack of thoroughness on my part. I will update the installation instructions for this too.

Once you try this, let me know if there are still any issues and we can go from there!

Thanks,
Vishal

rickstaa, 2019-06-24:
@visatish Thanks a lot for your answer. I will do some testing this afternoon and report the result. What do you use internally to compute the grasps and control the robot? My plan was now to build a C++ ROS node that spawns your python 3 detection algorithm as a new process and communicate through a shared memory address or UTP sockets.

visatish, 2019-06-24:
@rickstaa Internally we simply query the Python policy. The returned actions have a property called `grasp` which has a function `pose()` that returns the grasp pose in 3D camera space. You will need to transform to 3D robot space using a known transformation that can be calibrated with a chessboard. So you would execute something like this in code:

```
T_grasp_camera = action.grasp.pose()
T_grasp_robot = T_camera_robot * T_grasp_camera
```

For controlling the robot, we use an ABB YuMi and have an open-source Python library [here](https://github.com/BerkeleyAutomation/yumipy). You will probably have to find something similar for the Panda. Unfortunately, we don't have the bandwidth to help with motion planning itself.

rickstaa, 2019-06-26:
@visatish I did some testing within a python2.7 virtual environment. Ilke you stated your ROS package seems to work inside a python2.7 virtual environment. While following your steps I, however, encountered some errors. For documentation purposes I reported the errors and the solutions to these errors below:

# Problems while building the catkin package

## No module named em
```
Errors     << gqcnn:make /catkin_ws/logs/gqcnn/build.make.000.log                                                                                                                                                 
Traceback (most recent call last):
  File "/opt/ros/kinetic/share/gencpp/cmake/../../../lib/gencpp/gen_cpp.py", line 41, in <module>
    import genmsg.template_tools
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/genmsg/template_tools.py", line 39, in <module>
    import em
ImportError: No module named 'em'
make[2]: *** [/catkin_ws/devel/.private/gqcnn/include/gqcnn/BoundingBox.h] Error 1
make[2]: *** Waiting for unfinished jobs....
Traceback (most recent call last):
  File "/opt/ros/kinetic/share/gencpp/cmake/../../../lib/gencpp/gen_cpp.py", line 41, in <module>
    import genmsg.template_tools
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/genmsg/template_tools.py", line 39, in <module>
    import em
ImportError: No module named 'em'
make[2]: *** [/catkin_ws/devel/.private/gqcnn/include/gqcnn/Action.h] Error 1
Traceback (most recent call last):
  File "/opt/ros/kinetic/share/gencpp/cmake/../../../lib/gencpp/gen_cpp.py", line 41, in <module>
    import genmsg.template_tools
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/genmsg/template_tools.py", line 39, in <module>
    import em
ImportError: No module named 'em'
make[2]: *** [/catkin_ws/devel/.private/gqcnn/include/gqcnn/Observation.h] Error 1
make[1]: *** [CMakeFiles/gqcnn_generate_messages_cpp.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make: *** [all] Error 2
```

### Solution
Solved by running `pip install empy`

# Problems while running the ros node

## Wrong PYTHONPATH error
After sourcing the ROS `setup.bash` file and launching the ROS package using the `roslaunch gqcnn grasp_planning_service.launch` command I get the following errors:

```
Failed to load Python extension for LZ4 support. LZ4 compression will not be available.
Traceback (most recent call last):
  File "/catkin_ws/src/gqcnn/ros_nodes/grasp_planner_node.py", line 46, in <module>
    from perception import (CameraIntrinsics, ColorImage, DepthImage, BinaryImage,
  File "/catkin_ws/devel/lib/python3/dist-packages/perception/__init__.py", line 35, in <module>
    exec(__fh.read())
  File "<string>", line 8, in <module>
  File "/catkin_ws/src/perception/perception/camera_intrinsics.py", line 14, in <module>
    from .image import DepthImage, PointCloudImage
  File "/catkin_ws/src/perception/perception/image.py", line 11, in <module>
    import cv2
ImportError: /opt/ros/kinetic/lib/python2.7/dist-packages/cv2.so: undefined symbol: PyCObject_Type
[gqcnn/grasp_planner-2] process has died [pid 11905, exit code 1, cmd /catkin_ws/src/gqcnn/ros_nodes/grasp_planner_node.py __name:=grasp_planner __log:=/root/.ros/log/b6dee622-9815-11e9-a237-98e7f48e3bc4/gqcnn-grasp_planner-2.log].
log file: /root/.ros/log/b6dee622-9815-11e9-a237-98e7f48e3bc4/gqcnn-grasp_planner-2*.log
```

```
WARNING:root:autolab_core not installed as catkin package, RigidTransform ros methods will be unavailable
Traceback (most recent call last):
  File "/catkin_ws/src/gqcnn/ros_nodes/grasp_planner_node.py", line 46, in <module>
    from perception import (CameraIntrinsics, ColorImage, DepthImage, BinaryImage,
  File "build/bdist.linux-x86_64/egg/perception/__init__.py", line 8, in <module>
    
  File "build/bdist.linux-x86_64/egg/perception/camera_intrinsics.py", line 14, in <module>
  File "build/bdist.linux-x86_64/egg/perception/image.py", line 27, in <module>
  File "/usr/local/lib/python2.7/dist-packages/skimage/__init__.py", line 167, in <module>
    from .util.dtype import (img_as_float32,
[... 28 lines cut ...]

## NameError: global name 'FileNotFoundError' is not defined

```
Traceback (most recent call last):
  File "/catkin_ws/src/gqcnn/ros_nodes/grasp_planner_node.py", line 410, in <module>
    grasping_policy = CrossEntropyRobustGraspingPolicy(policy_cfg)
  File "/catkin_ws/src/gqcnn/gqcnn/grasping/policy/policy.py", line 726, in __init__
    GraspingPolicy.__init__(self, config)
  File "/catkin_ws/src/gqcnn/gqcnn/grasping/policy/policy.py", line 310, in __init__
    metric_type, self._metric_config)
  File "/catkin_ws/src/gqcnn/gqcnn/grasping/grasp_quality_function.py", line 1303, in quality_function
    return GQCnnQualityFunction(config)
  File "/catkin_ws/src/gqcnn/gqcnn/grasping/grasp_quality_function.py", line 931, in __init__
    self._gqcnn = get_gqcnn_model().load(self._gqcnn_model_dir)
  File "/catkin_ws/src/gqcnn/gqcnn/model/tf/network_tf.py", line 218, in load
    gqcnn.init_mean_and_std(model_dir)
  File "/catkin_ws/src/gqcnn/gqcnn/model/tf/network_tf.py", line 245, in init_mean_and_std
    except FileNotFoundError:
NameError: global name 'FileNotFoundError' is not defined
```

### Solution

This error is not present when I supply the `model_name:=GQCNN-4.0-PJ` argument with my roslaunch command.

closed 2019-06-26 by rickstaa