BerkeleyAutomation/gqcnn #78: ROS service python2 incompatibility

rickstaa, 2019-08-27:
**System information**
- OS Platform and Distribution: Ubuntu 18.04
- Python version: v2.7
- Installed using pip or ROS: ROS
- Camera: Kinect v2
- Gripper:  ParralelYaw
- Robot: Panda Emika Franka
- GPU model (if applicable): Nvidia M1000M GPU

**Describe what you are trying to do**
I'm trying to use your `Grasp_Sampler_Server` for computing my robot grasps.

**Describe current behavior**

Currently, I get the following error:

```
Traceback (most recent call last):
  File "/home/ricks/panda_autograsp_ws/src/gqcnn/ros_nodes/grasp_planner_node.py", line 410, in <module>
    grasping_policy = CrossEntropyRobustGraspingPolicy(policy_cfg)
  File "/home/ricks/panda_autograsp_ws/src/gqcnn/gqcnn/grasping/policy/policy.py", line 726, in __init__
    GraspingPolicy.__init__(self, config)
  File "/home/ricks/panda_autograsp_ws/src/gqcnn/gqcnn/grasping/policy/policy.py", line 310, in __init__
    metric_type, self._metric_config)
  File "/home/ricks/panda_autograsp_ws/src/gqcnn/gqcnn/grasping/grasp_quality_function.py", line 1303, in quality_function
    return GQCnnQualityFunction(config)
  File "/home/ricks/panda_autograsp_ws/src/gqcnn/gqcnn/grasping/grasp_quality_function.py", line 931, in __init__
    self._gqcnn = get_gqcnn_model().load(self._gqcnn_model_dir)
  File "/home/ricks/panda_autograsp_ws/src/gqcnn/gqcnn/model/tf/network_tf.py", line 218, in load
    gqcnn.init_mean_and_std(model_dir)
  File "/home/ricks/panda_autograsp_ws/src/gqcnn/gqcnn/model/tf/network_tf.py", line 245, in init_mean_and_std
    except FileNotFoundError:
NameError: global name 'FileNotFoundError' is not defined
```

**Describe the expected behaviour** 
I want to have no errors.

**Describe the input images**
Images of the `sensor_msg.Image` create by the [iai_kinectv2 toolbox](https://github.com/code-iai/iai_kinect2)

**Solution**
This is caused because FileNoteFound [was released in python3.3](https://stackoverflow.com/questions/28633555/how-to-handle-filenotfounderror-when-try-except-ioerror-does-not-catch-it). I created a version for myself in which the error is solved so I can use the toolbox with python 2.7 in ros Melodic till a ROS version is out that fully supports python3. This Version can be found in pull request #.

LHJ1098826475, 2019-10-14:
FileNotFoundError is the attribute of python3, python2.7 need to use IOError.

rickstaa, 2019-10-14:
@LHJ1098826475 Thanks for your response. That's I made a commit for people to use when they need to use the package with python 2.7 instead of python 3 (see [rickstaa/gqcnn/commit/094109688c3178b071418e6c0e0b3dc76564ab50](https://github.com/rickstaa/gqcnn/commit/094109688c3178b071418e6c0e0b3dc76564ab50)). As the IOError is both python2 and python3 compatible I also created a pull request might the maintainers want to merge it (see #80).

baihaisheng, 2019-11-28:
> @LHJ1098826475 Thanks for your response. That's indeed correct. I made a commit might people need to use the package with python 2.7 instead of python 3 (see [rickstaa/gqcnn/commit/094109688c3178b071418e6c0e0b3dc76564ab50](https://github.com/rickstaa/gqcnn/commit/094109688c3178b071418e6c0e0b3dc76564ab50)). As using IOError is to my knowledge both python2 and python3 compatible I also created a pull request might the maintainers want to merge it (see #80).

Hi,@rickstaa
I had some problems while following the tutorial ROS Installation.
Can I refer you to the instruction steps during installation.
Thank you.

rickstaa, 2019-11-28:
Hi @baihaisheng I am not entirely sure If I understand you. Do you have a problem with the pull request I created or the master branch of the GQCNN package?

baihaisheng, 2019-11-29:
> Hi @baihaisheng. I am not entirely sure If I understand you. Do you have a problem with the pull request I created or the master branch of the GQCNN package?

Hi,@rickstaa. I also have the same problem when I  use the ROS Installation,  Roslaunch not working, I use the python2.7, how did you solve it?

rickstaa, 2019-11-29:
Ah, I see! A small fix to the problem you are experiencing can be found in my fork of the gqcnn repository. The pull request of this fix can be found [here](https://github.com/BerkeleyAutomation/gqcnn/pull/80). Until it is reviewed by @visatish or any of the other BerkeleyAutomation department members, I think it is easiest to cherry-pick my commit. This can be done by using the following commands from within your local gqcnn repository folder:

```
git remote add rstaa git@github.com:rickstaa/gqcnn.git
git fetch rstaa
git cherry-pick 094109688c3178b071418e6c0e0b3dc76564ab50
```

Let me know if that solved the problems your experiencing.

baihaisheng, 2019-12-01:
> git cherry-pick 094109688c3178b071418e6c0e0b3dc76564ab50
Hi,@rickstaa
when I carry out the git fetch rstaa,it appeared a problem
[image]
when I use the roslaunch gqcnn grasp_planning_service.launch model_name:=GQCNN-4.0-PJ,it shows as follow:
[image]
so I try to run the pip install .,but is also cann't work
[image]
can you tell me what can I do to solve this,thank you.

rickstaa, 2019-12-01:
@baihaisheng The error you are getting is because `ruamel.yaml` is missing in your python environment. Did you use the `pip install .` command inside the main gqcnn folder? This should install the `ruamel.yaml` package which is need by the `autolab_core` package. Just an heads up, @visatish just created another pull request with a fix to the problem you are experiencing. I would therefore now advise you to use his commit instead of mine since it is a bit stricter compared to the generic IOError which I used.

rickstaa, 2019-12-01:
@baihaisheng Just another heads up. The fix by @visatish was just merged with the master.

visatish, 2019-12-01:
Hi @baihaisheng,

As @rickstaa mentioned, please use the master branch going forward.

Regarding your issue with `libspatialindex_c.so`, it appears that the latest version of `rtree` requires `libspatialindex` to be installed. Please see lines 58-63 in the [Travis build config](https://github.com/BerkeleyAutomation/gqcnn/blob/master/.travis.yml#L58) for how to install it.

Let me know if you run into any more issues.

Thanks,
Vishal

baihaisheng, 2019-12-02:
> Hi @baihaisheng,
>
> As @rickstaa mentioned, please use the master branch going forward.
>
> Regarding your issue with `libspatialindex_c.so`, it appears that the latest version of `rtree` requires [...]

Hi @visatish @rickstaa 
Thank you for your quick answers,I installed the libspatialindex,and run pip install .
the roslaunch command can work now,and I think the ROS installation tutorials can be more detailed.
Thanks
baihaisheng

visatish, 2019-12-02:
Hi @baihaisheng,

Glad to hear that you were able to get things up and running! Apologies for the bumpy road, I will improve the ROS installation tutorial when I get a chance!

Thanks,
Vishal

baihaisheng, 2020-07-25:
> Hi @baihaisheng,
> 
> Glad to hear that you were able to get things up and running! Apologies for the bumpy road, I will improve the ROS installation tutorial when I get a chance!
> 
> Thanks,
> Vishal

Hi @visatish
I want to fine-tune the weights of a pre-trained Dex-Net 4.0 GQ-CNN model, but I can't find the training dataset under the path to the training dataset as follows:

_$ python tools/finetune.py data/training/example_pj/ GQCNN-4.0-PJ --config_filename cfg/finetune_example_pj.yaml --name gqcnn_example_pj_

**There is no training file in the directory data**。

visatish, 2020-07-27:
Hi @baihaisheng,

The example fine-tuning datasets can be downloaded with `scripts/downloads/download_example_data.sh`. It should be run from the top-most project dir.

Thanks,
Vishal

baihaisheng, 2020-07-29:
> Hi @baihaisheng,
> 
> The example fine-tuning datasets can be downloaded with `scripts/downloads/download_example_data.sh`. It should be run from the top-most project dir.
> 
> Thanks,
> Vishal

Hi @visatish
Thank you for your quick answer.

closed 2020-07-30 by rickstaa