BerkeleyAutomation/gqcnn #81: ROS in Docker Installation

SimonePanicucci, 2019-09-16:
**System information**
- Linux Ubuntu 16.04
- Docker 19.03.1
- Cuda 10.0

Hi, I found docker installation really useful and effectively, however, I'm wondering if it's possible to have gqcnn with ROS in Docker version.
I'm asking this since I would like to test the solution on a physical robot using a Kinect v2.
Many thanks and have a nice day.

visatish, 2019-09-16:
Hi @SimonePanicucci ,

Glad to hear you found the Docker installation useful! Unfortunately, we do not have a version of the Docker installation for ROS, and I don't have the bandwidth right now to add that. However, you could try implementing it yourself and then possibly make a PR-it shouldn't be too bad. All you will have to do is edit the appropriate Docker file and 1) Add ROS along with any deps, then 2) Replace the normal installation commands with the ROS installation commands from our docs [here](https://berkeleyautomation.github.io/gqcnn/install/install.html#ros-installation).

I can provide some guidance if you choose to do this. Also, what is your physical robot setup like? Our pre-trained networks are designed for particular camera, gripper, and workspace setups with some tolerance for variation, but we can't make any performance guarantees.

Also cc'ing @jeffmahler. 

Thanks,
Vishal

SimonePanicucci, 2019-09-19:
Hi @visatish,
many thanks for your instructions, I think I'll try on my own.

About the setup, I'm using a Kinect2 over a table 70cm far, I'm working on use that as input and to modify the code. In perception module I saw something about the Kinect2, it should't be too tricky.
Am I right?

Many thanks,
Simone

closed 2019-09-24 by SimonePanicucci