BerkeleyAutomation/gqcnn #85: License question

rickstaa, 2019-11-27:
I created an autonomous grasping solution for the kinect v2 and the Panda Emika Franka robot using your pretrained GQ-CNN model. And I have to say that it performs reasonably well considering it was not retrained for my specific setup. The code of this solution can be found [here](https://github.com/rickstaa/panda_autograsp). I just wanted to do a quick check whether I am allowed to publish my grasping solution which uses your pre-trained CNN the way It is currently published. I published my code under an MIT licence and added note that the license of the submodules, including yours should be respected.

visatish, 2019-11-27:
Hi @rickstaa,

Wow, this is really cool-amazing work! I think this is the first time someone has developed another open-source project utilizing GQ-CNNs so I'm excited 🎉. 

I believe this is perfectly fine, however I'm going to cc @jeffmahler since he has far more experience on this sort of thing.

Thanks,
Vishal

rickstaa, 2019-11-29:
@visatish Thanks a lot! I am impressed with how good your pre-trained GQ-CNN and FC-CNN's work with my setup which uses a Kinect instead of a Photoneo PhoXi S sensor. I am planning to retrain them in the future after adding some additional grasping methods.

visatish, 2019-11-29:
@rickstaa Glad to hear that and looking forward to your future plans!

baihaisheng, 2019-12-05:
> I created an autonomous grasping solution for the kinect v2 and the Panda Emika Franka robot using your pretrained GQ-CNN model. And I have to say that it performs reasonably well considering it was not retrained for my specific setup. The code of this solution can be found [here](https://github.com/rickstaa/panda_autograsp). I just wanted to do a quick check whether I am allowed to publish my grasping solution which uses your pre-trained CNN the way It is currently published. I published my code under an MIT licence and added a note that the license of the submodules, including yours should be respected.

Hi @rickstaa ,I am using the kinectV2 to test the GQ-CNN4.0-PJ,but I don't know how to connect my camera with the gqcnn package,I want to have such depth_3.npy and segmask_3.png,could tell me how to do this?
Thank you very much.

rickstaa, 2019-12-06:
Hey @baihaisheng, Of course, I created the camera connection by using the [IAI_kinect2 ROS package](https://github.com/code-iai/iai_kinect2) while modifying the original GQCNN python ROS node script. You can take a lock at the repository to see how this is done. I did not include a segmask yet.

baihaisheng, 2019-12-09:
> Hey @baihaisheng, Of course, I created the camera connection by using the [IAI_kinect2 ROS package](https://github.com/code-iai/iai_kinect2) while modifying the original GQCNN python ROS node script. You can take a lock at the repository to see how this is done. I did not include a segmask yet. The repository can be found [here](https://github.com/rickstaa/panda_autograsp).

Hi, @rickstaa 
Thank you for your quick reply, I will take a good look at your repository. By the way, did you modify the _example/policy_ros.py_? I just don't know how to modify this with my kinect2, could you help me ?
Thanks.

rickstaa, 2019-12-17:
Hi, @baihaisheng.

To connect the GQCNN package to the Kinect camera. I first start the [IAI_kinect2](https://github.com/code-iai/iai_kinect2) processing node using the following code:

https://github.com/rickstaa/panda_autograsp/blob/85d021857b69f3c307fed95ecb42a7d90b35ebbf/panda_autograsp/launch/panda_autograsp.launch#L79-L84

I then launch my modified [grasp_planner_node.py](https://github.com/rickstaa/panda_autograsp/blob/melodic-devel/panda_autograsp/nodes/grasp_planner_server.py) together with my [panda_autograsp_server_node](https://github.com/rickstaa/panda_autograsp/blob/melodic-devel/panda_autograsp/nodes/panda_autograsp_server.py). In this node, I then subscribe to the camera topics that are created by the [IAI_kinect2](https://github.com/code-iai/iai_kinect2)  processing node:

https://github.com/rickstaa/panda_autograsp/blob/85d021857b69f3c307fed95ecb42a7d90b35ebbf/panda_autograsp/src/panda_autograsp/panda_autograsp_server_ros.py#L491-L533

Lastly, I call the `grasp_planner` service using the camera messages I received from the IAI_kinect2 messages:

https://github.com/rickstaa/panda_autograsp/blob/85d021857b69f3c307fed95ecb42a7d90b35ebbf/panda_autograsp/src/panda_autograsp/panda_autograsp_server_ros.py#L616-L627

Hope that helps let me know if you run into problems.

baihaisheng, 2019-12-17:
> Hi, @baihaisheng.
> 
> To connect the GQCNN package to the Kinect camera. I first start the [IAI_kinect2](https://github.com/code-iai/iai_kinect2) processing node using the following code:
> 
> [panda_autograsp.launch#L79-L84](https://github.com/rickstaa/panda_autograsp/blob/85d021857b69f3c307fed95ecb42a7d90b35ebbf/panda_autograsp/launch/panda_autograsp.launch#L79-L84)
> 
> I then launch my modified [grasp_planner_node.py](https://github.com/rickstaa/panda_autograsp/blob/melodic-devel/panda_autograsp/nodes/grasp_planner_server.py) together with my [panda_autograsp_server_node](https://github.com/rickstaa/panda_autograsp/blob/melodic-devel/panda_autograsp/nodes/panda_autograsp_server.py). In this node, I then subscribe to the camera topics that are created by the [IAI_kinect2](https://github.com/code-iai/iai_kinect2) processing node:
> 
> [panda_autograsp_server_ros.py#L491-L533](https://github.com/rickstaa/panda_autograsp/blob/85d021857b69f3c307fed95ecb42a7d90b35ebbf/panda_autograsp/src/panda_autograsp/panda_autograsp_server_ros.py#L491-L533)
> 
> Lastly, I call the `grasp_planner` service using the camera messages I received from the IAI_kinect2 messages:
> 
> [panda_autograsp_server_ros.py#L616-L627](https://github.com/rickstaa/panda_autograsp/blob/85d021857b69f3c307fed95ecb42a7d90b35ebbf/panda_autograsp/src/panda_autograsp/panda_autograsp_server_ros.py#L616-L627)
> 
> Hope that helps let me know if you run into problems.

Hi, @rickstaa 
Thank you so much for your reply, now I can test my own depth_image with GQCNN-4.0-PJ， and
I got the pose and my commands are as follows:
[image]
$ roslaunch gqcnn grasp_planning_service.launch model_name:=GQCNN-4.0-PJ
$ python examples/policy_ros.py --depth_image data/examples/clutter/phoxi/dex-net_4.0/depth_3.npy --segmask data/examples/clutter/phoxi/dex-net_4.0/segmask_3.png --camera_intr data/calib/phoxi/phoxi.intr
Now I want to test the FC-GQCNN-4.0-PJ in ROS, which commands can use?
Thanks.

visatish, 2019-12-18:
@baihaisheng Please see https://berkeleyautomation.github.io/gqcnn/tutorials/tutorial.html#with-ros.

baihaisheng, 2019-12-26:
> @baihaisheng Please see https://berkeleyautomation.github.io/gqcnn/tutorials/tutorial.html#with-ros.

Hi, @visatish 
I observed that the gqcnn package was modified, do I need to run the git pull command to synchronize?

baihaisheng, 2019-12-26:
> @baihaisheng Please see https://berkeleyautomation.github.io/gqcnn/tutorials/tutorial.html#with-ros.

Hi, @visatish 
I run the _git pull_ command to update gqcnn package, but when I run the roslaunch command, something went wrong. 

$ roslaunch gqcnn grasp_planning_service.launch model_name:=GQCNN-4.0-PJ

[image]

Then I try to run the 
$ roslaunch gqcnn grasp_planning_service.launch model_name:=FC-GQCNN-4.0-PJ fully_conv:=True
Nothing was wrong.

visatish, 2019-12-26:
Hi @baihaisheng,

Sorry about that, a fix has been pushed to master!

Thanks,
Vishal

baihaisheng, 2019-12-30:
> Hi @baihaisheng,
> 
> Sorry about that, fixed in #96!
> 
> Thanks,
> Vishal

Hi, @visatish @jeffmahler @rickstaa 
I have finished the ROS tutorial, now I can get the grasp_pose showed in the image just like this
[image]
the command that I run as follows:
$ roslaunch gqcnn grasp_planning_service.launch model_name:=GQCNN-4.0-PJ
$ python examples/policy_ros.py --depth_image data/examples/clutter/phoxi/dex-net_4.0/depth_3.npy --segmask data/examples/clutter/phoxi/dex-net_4.0/segmask_3.png --camera_intr data/calib/phoxi/phoxi.intr
Then I want to control the UR5 robot to grasp the object, so I run the command _rostopic list_ and I find the _/gqcnn_grasp/pose_, I run the _rostopic echo /gqcnn_grasp/pose_ and I got the pose.
[image]
[image]

but I find that the orientation maybe not right, I want to know the object coordinate. Can you help me to answer this, thank you so much.

rickstaa, 2020-01-23:
Dear @baihaisheng Thanks a lot for using my package. The full documentation to the package can be found [here](https://rickstaa.github.io/panda_autograsp/). The package can be started using the following command:

```
roslaunch panda_autograsp panda_autograsp.launch
```

closed 2020-01-23 by rickstaa

sunsetme, 2020-09-14:
hello! I am new to ROS,