BerkeleyAutomation/gqcnn #97: meshrender installation error

fellenB, 2020-01-09:
**System information**
- OS :  Ubuntu 16.04
- Python version: python3.5
- Installed using pip 

**Describe the problem**
virtualization package depend on meshrender(version 0.0.10), there is something wrong with the installation of meshrender. 

**The installation log**
Collecting visualization
  Using cached https://files.pythonhosted.org/packages/05/6e/3637994716570f6a727075eab2a346142775b44d919b2a0a1ecf6e0a0208/visualization-0.0.6.tar.gz
Collecting imageio
  Using cached https://files.pythonhosted.org/packages/1a/de/f7f985018f462ceeffada7f6e609919fbcc934acd9301929cba14bc2c24a/imageio-2.6.1-py3-none-any.whl
Requirement already satisfied: numpy in /home/siasun/.virtualenvs/gqcnn_python3.5/lib/python3.5/site-packages (from visualization) (1.18.0)
Requirement already satisfied: matplotlib<=2.2.0 in /home/siasun/.virtualenvs/gqcnn_python3.5/lib/python3.5/site-packages (from visualization) (2.2.0)
Collecting trimesh[easy]
  Using cached https://files.pythonhosted.org/packages/8e/50/182e365ab5f3b715ebe5e11bad7386fd2a86a36de0c1067a4ea6ec906e1d/trimesh-3.5.13.tar.gz
Requirement already satisfied: autolab_core in /home/siasun/.virtualenvs/gqcnn_python3.5/lib/python3.5/site-packages (from visualization) (0.0.13)
Requirement already satisfied: autolab_perception in /home/siasun/.virtualenvs/gqcnn_python3.5/lib/python3.5/site-packages (from visualization) (0.0.8)
Collecting meshrender
  Using cached https://files.pythonhosted.org/packages/00/0a/3133ecb4b885b33d21531645dda39b2c62d9f0683582fe002adacaf2d257/meshrender-0.0.10.tar.gz
    ERROR: Command errored out with exit status 1:
     command: /home/siasun/.virtualenvs/gqcnn_python3.5/bin/python3.5 -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-v5d595zn/meshrender/setup.py'"'"'; __file__='"'"'/tmp/pip-install-v5d595zn/meshrender/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base /tmp/pip-install-v5d595zn/meshrender/pip-egg-info
         cwd: /tmp/pip-install-v5d595zn/meshrender/
    Complete output (51 lines):
        ERROR: Command errored out with exit status 1:
         command: /home/siasun/.virtualenvs/gqcnn_python3.5/bin/python3.5 -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-wheel-xpvy956a/rtree/setup.py'"'"'; __file__='"'"'/tmp/pip-wheel-xpvy956a/rtree/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base /tmp/pip-wheel-xpvy956a/rtree/pip-egg-info
             cwd: /tmp/pip-wheel-xpvy956a/rtree/
        Complete output (15 lines):
        Traceback (most recent call last):
          File "<string>", line 1, in <module>
          File "/tmp/pip-wheel-xpvy956a/rtree/setup.py", line 3, in <module>
            import rtree
          File "/tmp/pip-wheel-xpvy956a/rtree/rtree/__init__.py", line 1, in <module>
            from .index import Rtree
          File "/tmp/pip-wheel-xpvy956a/rtree/rtree/index.py", line 6, in <module>
            from . import core
          File "/tmp/pip-wheel-xpvy956a/rtree/rtree/core.py", line 145, in <module>
            rt.Error_GetLastErrorNum.restype = ctypes.c_int
          File "/usr/lib/python3.5/ctypes/__init__.py", line 360, in __getattr__
            func = self.__getitem__(name)
          File "/usr/lib/python3.5/ctypes/__init__.py", line 365, in __getitem__
            func = self._FuncPtr((name_or_ordinal, self))
        AttributeError: /home/siasun/.virtualenvs/gqcnn_python3.5/bin/python3.5: undefined symbol: Error_GetLastErrorNum
        ----------------------------------------
    ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.
    Traceback (most recent call last):
      File "/home/siasun/.virtualenvs/gqcnn_python3.5/lib/python3.5/site-packages/setuptools/installer.py", line 128, in fetch_build_egg
        subprocess.check_call(cmd)
      File "/usr/lib/python3.5/subprocess.py", line 581, in check_call
        raise CalledProcessError(retcode, cmd)
    subprocess.CalledProcessError: Command '['/home/siasun/.virtualenvs/gqcnn_python3.5/bin/python3.5', '-m', 'pip', '--disable-pip-version-check', 'wheel', '--no-deps', '-w', '/tmp/tmp1h7m77zq', '--quiet', 'rtree']' returned non-zero exit status 1
    
    During handling of the above exception, another exception occurred:
    
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-install-v5d595zn/meshrender/setup.py", line 47, in <module>
        'sphinx_rtd_theme'
      File "/home/siasun/.virtualenvs/gqcnn_python3.5/lib/python3.5/site-packages/setuptools/__init__.py", line 144, in setup
        _install_setup_requires(attrs)
      File "/home/siasun/.virtualenvs/gqcnn_python3.5/lib/python3.5/site-packages/setuptools/__init__.py", line 139, in _install_setup_requires
        dist.fetch_build_eggs(dist.setup_requires)
      File "/home/siasun/.virtualenvs/gqcnn_python3.5/lib/python3.5/site-packages/setuptools/dist.py", line 721, in fetch_build_eggs
        replace_conflicting=True,
      File "/home/siasun/.virtualenvs/gqcnn_python3.5/lib/python3.5/site-packages/pkg_resources/__init__.py", line 782, in resolve
        replace_conflicting=replace_conflicting
      File "/home/siasun/.virtualenvs/gqcnn_python3.5/lib/python3.5/site-packages/pkg_resources/__init__.py", line 1065, in best_match
        return self.obtain(req, installer)
      File "/home/siasun/.virtualenvs/gqcnn_python3.5/lib/python3.5/site-packages/pkg_resources/__init__.py", line 1077, in obtain
        return installer(requirement)
      File "/home/siasun/.virtualenvs/gqcnn_python3.5/lib/python3.5/site-packages/setuptools/dist.py", line 777, in fetch_build_egg
        return fetch_build_egg(self, req)
      File "/home/siasun/.virtualenvs/gqcnn_python3.5/lib/python3.5/site-packages/setuptools/installer.py", line 130, in fetch_build_egg
        raise DistutilsError(str(e))
    distutils.errors.DistutilsError: Command '['/home/siasun/.virtualenvs/gqcnn_python3.5/bin/python3.5', '-m', 'pip', '--disable-pip-version-check', 'wheel', '--no-deps', '-w', '/tmp/tmp1h7m77zq', '--quiet', 'rtree']' returned non-zero exit status 1
    ----------------------------------------
ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.

visatish, 2020-01-09:
Hi @fellenB,

This seems to be an issue with rtree. Can you try uninstalling/reinstalling rtree alone and make sure `import rtree` works?

Another possibility is to try a different Python version if possible. This shouldn't really matter, but it's worth a try.

You could also try building the provided dockerfiles instead as instructed [here](https://berkeleyautomation.github.io/gqcnn/install/install.html#docker-installation).

The [CI builds](https://travis-ci.org/BerkeleyAutomation/gqcnn/) do seem to be passing, even for Python3.5.

Thanks,
Vishal

fellenB, 2020-01-13:
thanks for your response! the rtree issue is corrected referring to [https://github.com/Toblerity/rtree/issues/120.](url)
but the there is also something wrong with installation of meshrender.
__________________________________________________________________________
pip install visualization --default-time=10000
Collecting visualization
  Using cached https://files.pythonhosted.org/packages/05/6e/3637994716570f6a727075eab2a346142775b44d919b2a0a1ecf6e0a0208/visualization-0.0.6.tar.gz
Collecting imageio
  Using cached https://files.pythonhosted.org/packages/1a/de/f7f985018f462ceeffada7f6e609919fbcc934acd9301929cba14bc2c24a/imageio-2.6.1-py3-none-any.whl
Requirement already satisfied: numpy in /home/siasun/.virtualenvs/gqcnn_python3.5/lib/python3.5/site-packages (from visualization) (1.18.0)
Requirement already satisfied: matplotlib<=2.2.0 in /home/siasun/.virtualenvs/gqcnn_python3.5/lib/python3.5/site-packages (from visualization) (2.2.0)
Collecting trimesh[easy]
  Using cached https://files.pythonhosted.org/packages/bd/10/4d4e0b06a94008953f0d6d73a94904761820075022082b8a72ff16e70763/trimesh-3.5.14.tar.gz
Requirement already satisfied: autolab_core in /home/siasun/.virtualenvs/gqcnn_python3.5/lib/python3.5/site-packages (from visualization) (0.0.13)
Requirement already satisfied: autolab_perception in /home/siasun/.virtualenvs/gqcnn_python3.5/lib/python3.5/site-packages (from visualization) (0.0.8)
Collecting meshrender
  Using cached https://files.pythonhosted.org/packages/00/0a/3133ecb4b885b33d21531645dda39b2c62d9f0683582fe002adacaf2d257/meshrender-0.0.10.tar.gz
    ERROR: Command errored out with exit status 1:
     command: /home/siasun/.virtualenvs/gqcnn_python3.5/bin/python3.5 -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-wq2codk5/meshrender/setup.py'"'"'; __file__='"'"'/tmp/pip-install-wq2codk5/meshrender/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base /tmp/pip-install-wq2codk5/meshrender/pip-egg-info
         cwd: /tmp/pip-install-wq2codk5/meshrender/
    Complete output (91 lines):
      WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.VerifiedHTTPSConnection object at 0x7fb2fcc17908>: Failed to establish a new connection: [Errno 101] Network is unreachable',)': /packages/a5/fa/5bd627f889327999736350637d7e759a17486d32b69be0c703f7abc64e4e/msgpack-0.6.2-cp35-cp35m-manylinux1_x86_64.whl
    ERROR: Exception:
    Traceback (most recent call last):
      File "/home/siasun/.virtualenvs/gqcnn_python3.5/lib/python3.5/site-packages/pip/_vendor/urllib3/response.py", line 425, in _error_catcher
        yield
      File "/home/siasun/.virtualenvs/gqcnn_python3.5/lib/python3.5/site-packages/pip/_vendor/urllib3/response.py", line 507, in read
        data = self._fp.read(amt) if not fp_closed else b""
      File "/home/siasun/.virtualenvs/gqcnn_python3.5/lib/python3.5/site-packages/pip/_vendor/cachecontrol/filewrapper.py", line 62, in read
        data = self.__fp.read(amt)
      File "/usr/lib/python3.5/http/client.py", line 458, in read
        n = self.readinto(b)
      File "/usr/lib/python3.5/http/client.py", line 498, in readinto
        n = self.fp.readinto(b)
      File "/usr/lib/python3.5/socket.py", line 575, in readinto
        return self._sock.recv_into(b)
      File "/usr/lib/python3.5/ssl.py", line 929, in recv_into
        return self.read(nbytes, buffer)
      File "/usr/lib/python3.5/ssl.py", line 791, in read
        return self._sslobj.read(len, buffer)
      File "/usr/lib/python3.5/ssl.py", line 575, in read
        v = self._sslobj.read(len, buffer)
    socket.timeout: The read operation timed out
    
    During handling of the above exception, another exception occurred:
    
    Traceback (most recent call last):
      File "/home/siasun/.virtualenvs/gqcnn_python3.5/lib/python3.5/site-packages/pip/_internal/cli/base_command.py", line 153, in _main
        status = self.run(options, args)
      File "/home/siasun/.virtualenvs/gqcnn_python3.5/lib/python3.5/site-packages/pip/_internal/commands/wheel.py", line 158, in run
        resolver.resolve(requirement_set)
      File "/home/siasun/.virtualenvs/gqcnn_python3.5/lib/python3.5/site-packages/pip/_internal/legacy_resolve.py", line 201, in resolve
        self._resolve_one(requirement_set, req)
      File "/home/siasun/.virtualenvs/gqcnn_python3.5/lib/python3.5/site-packages/pip/_internal/legacy_resolve.py", line 365, in _resolve_one
        abstract_dist = self._get_abstract_dist_for(req_to_install)
      File "/home/siasun/.virtualenvs/gqcnn_python3.5/lib/python3.5/site-packages/pip/_internal/legacy_resolve.py", line 313, in _get_abstract_dist_for
        req, self.session, self.finder, self.require_hashes
      File "/home/siasun/.virtualenvs/gqcnn_python3.5/lib/python3.5/site-packages/pip/_internal/operations/prepare.py", line 194, in prepare_linked_requirement
        progress_bar=self.progress_bar
      File "/home/siasun/.virtualenvs/gqcnn_python3.5/lib/python3.5/site-packages/pip/_internal/download.py", line 465, in unpack_url
        progress_bar=progress_bar
      File "/home/siasun/.virtualenvs/gqcnn_python3.5/lib/python3.5/site-packages/pip/_internal/download.py", line 316, in unpack_http_url
        progress_bar)
[... cut ...]
      File "/home/siasun/.virtualenvs/gqcnn_python3.5/lib/python3.5/site-packages/setuptools/installer.py", line 130, in fetch_build_egg
        raise DistutilsError(str(e))
    distutils.errors.DistutilsError: Command '['/home/siasun/.virtualenvs/gqcnn_python3.5/bin/python3.5', '-m', 'pip', '--disable-pip-version-check', 'wheel', '--no-deps', '-w', '/tmp/tmptdtuptvz', '--quiet', 'msgpack']' returned non-zero exit status 2
    ----------------------------------------
ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.

visatish, 2020-01-13:
Hmm, this seems to be a network issue preventing pip from fetching the remote package:

`Failed to establish a new connection: [Errno 101] Network is unreachable`

The previous packages are all being installed from locally cached versions. Can you check if you have a working network connection?

fellenB, 2020-01-14:
That‘s right！The network is available to install other package such as rtree but is not available to install meshrender.  I change network connection and it works. Thank you very much!

closed 2020-01-14 by fellenB