BerkeleyAutomation/gqcnn #99: Issue: Bug/Performance Issue [Replication]

zuoym15, 2020-01-13:
**System information**
- OS Platform and Distribution (e.g., Linux Ubuntu 16.04): Linux Ubuntu 16.04
- Python version: 2.7
- Installed using pip or ROS: pip
- GPU model (if applicable):

**Describe the result you are trying to replicate**
If you can, provide a link to the section in the [documentation]
https://github.com/BerkeleyAutomation/gqcnn/blob/master/scripts/training/train_dex-net_2.0.sh

**Provide the exact sequence of commands / steps that you executed to replicate this result**

**Describe the unexpected behavior** 
I'm trying to replicate the result from the Dex-Net 2.0 paper and I have a few questions. I would be much appreciated if you could help me. 

I ran the code scripts/training/train_dex-net_2.0.sh. I got a final validation error of 12.239% with the precision-recall and roc curve attached. Is that what I expect to see? I'm confused because in the original paper you report an 85.7% accuracy on the validation set and a 92.2% accuracy here (https://berkeleyautomation.github.io/gqcnn/benchmarks/benchmarks.html). My result equals to neither of these so I would like to know what result should I expect?

Thank you in advance for your help and I look forward to hearing from you!

[image]
[image]

JohnsonQi, 2020-03-17:
I have the same problem.

visatish, 2020-03-17:
Resolved over email.

closed 2020-03-17 by visatish