BerkeleyAutomation/gqcnn #110: Issue: Installation Issue

nalindas9, 2020-03-26:
**System information**
- OS Platform and Distribution (e.g., Linux Ubuntu 16.04): Linux Ubuntu 16.04
- Python version: 2.7.12 (virtualenv)
- ROS version: Kinetic
- Installed using pip or ROS: ROS

**Describe the problem**
I am trying to run the Dexnet 4.0 ROS Policy (https://berkeleyautomation.github.io/gqcnn/tutorials/tutorial.html#) in a Virtual Environment. The problem is that I am facing installation issues while installing the "Visualization" package from cpu_requirements.txt. All of the other packages are installing correctly. 

I tried installing the visualization package in python 3.7 in anaconda and it works. However, it is not able to install it for Python2. Therefore, it is not able to import the visualization module in "policy.py". I am using python 2.7.12 in my virtual environment (using Virtualenv) since I want to avoid conflicts with ROS. 
 
Could you let me know how to install the visualization package with pip on python2?
Thanks!

**Provide the exact sequence of commands / steps that you executed before running into the problem**
*Creating the virtual environment*
1. virtualenv -p /home/username/opt/python-2.7.15/bin/python dexnet

*ROS Installation*
1. Followed the same process given here - https://berkeleyautomation.github.io
/gqcnn/install/install.html#ros-installation 

*Installation of CPU and GPU requirements*
1. Installed each package separately using "pip install". For eg. pip install autolab_core. 
2. Facing problem while running "pip install visualization".  
The error log is attached.   
[error_log_visualization.txt](https://github.com/BerkeleyAutomation/gqcnn/files/4384342/error_log_visualization.txt)
3. Hence problem while running "roslaunch gqcnn grasp_planning_service.launch model_name:=GQCNN-4.0-PJ".

> Traceback (most recent call last):                                                            
  File "/home/nalindas9/sawyer_ws/src/gqcnn/ros_nodes/grasp_planner_node.py", line 49, in <mod
ule>                                                                                          
    from visualization import Visualizer2D as vis                                             
ImportError: No module named visualization                                                    

**Any other info / logs**

visatish, 2020-03-26:
Hi @nalindas9,

After a bit of googling around, it seems that your issue is caused by a missing `libspatialindex` as mentioned [here](https://github.com/Toblerity/rtree/issues/135). Can you try following [these](https://github.com/BerkeleyAutomation/gqcnn/blob/master/.travis.yml#L53) instructions from our Travis build to install it and then see if `pip install rtree` works?

Thanks,
Vishal

nalindas9, 2020-03-26:
@visatish Okay sure. I will try that. Thank you! Also, I tried installing the visualization package via pip as well as pip3 on other computers as well. It's throwing up the same error. I tried installing the previous releases too. Nothing worked.
I also tried updating setup tools using  

> pip install -U setuptools

It showed the same error. I think there is an issue in the visualization package. It seems to work only on python 3.7 +.

closed 2020-04-02 by nalindas9