BerkeleyAutomation/gqcnn #133: Issue: Installation Issue: [Cannot download pre-trained model and training datasets]

SonYeongGwang, 2022-11-03:
**System information**
- OS Platform and Distribution (e.g., Linux Ubuntu 16.04): Linux Ubuntu 16.04
- Python version: 3.5
- Installed using pip or ROS: pip

Hi, thank you for sharing and having us to have opportunity to refer such a great work.
I found that downloading the pre-trained model and datasets are no longer support in (https://berkeley.app.box.com/s/szbchyt3tou9e4ct6dz8c5v99vhx0s84)
I ask you to have alternative way to find the sources!

Thanks

timdreier, 2022-11-20:
@SonYeongGwang seems like the files are accessible again!

SonYeongGwang, 2022-11-20:
It seems like! Thank you for letting me know!

manish-kumar-01, 2022-12-12:
@SonYeongGwang Again, the trained models are inaccessible.

SonYeongGwang, 2022-12-27:
@manish-kumar-01 I guess they moved the like from 'box' to 'g-drive'. You can find the google drive directly linked to the dexnet models at https://berkeleyautomation.github.io/dex-net/

closed 2024-11-01 by SonYeongGwang